patience4711/read-APSystems-YC600-QS1-DS3 #4: compile problems

mamaretti, 2021-11-16:
hi patience4711

want to compile my own version with different pinouts

but im unable to compile it with IDE1.8.16 (mac)

i also changed some .H to lowercase

but now I can't solve this.
do you have an idea?

`/Users/xxx/arduino/ESP-ECU-v6_12/ESP-ECU-v6_12.ino: In function 'void loop()':
ESP-ECU-v6_12:399:7: error: 'class AsyncWebSocket' has no member named 'cleanupClients'
  399 |    ws.cleanupClients();
      |       ^~~~~~~~~~~~~~
/Users/xxx/arduino/ESP-ECU-v6_12/MQTT.ino: In function 'void MQTT_Receive_Callback(char*, byte*, unsigned int)':
MQTT:76:13: error: 'class ArduinoJson680_0_0::StaticJsonDocument<1024>' has no member named 'containsKey'
   76 |     if( doc.containsKey("poll") )
      |             ^~~~~~~~~~~
Mehrere Bibliotheken wurden für "WiFiUdp.h" gefunden
 Benutzt: /Users/xxx/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.2/libraries/ESP8266WiFi
 Nicht benutzt: /Applications/arduino/Arduino1816.app/Contents/Java/libraries/WiFi`

No13, 2021-11-17:
I tried to compile this code too, I needed to install the following libraries:
- ArduinoJson (Benoit Blanchon)
- AsyncHTTPRequest (Bob Lemaire,Khoi Hoang)
- ESPAsyncTCP (Me-No-Dev)
- ESPAsyncWebServer (Me-No-Dev)
- EspMQTTClient (Patrick Lapointe)
- NTPClient (Fabrice Weinberg)
- PubSubClient (Nick O'Leary)
- sunMoon (https://github.com/sfrwmaker/sunMoon)
- Time (Michael Margolis / Paul Stoffregen)

After this it seems to be working :)

adx-git, 2021-12-08:
Anyone got in building with Visual Studio Code?

closed 2021-12-30 by patience4711