patience4711/read-APSystems-YC600-QS1-DS3 #7: Nightmode triggered when poll reply is 'damaged'

No13, 2021-11-30:
Based on your work and the work of Kadzsol, I am working on a python library to communicate with the APS YC600 inverter.
While working on this I am using your firmware for ESP12 to monitor the current setup.

Today I managed to reproduce 'the nightmode bug' as discussed in this thread:
https://github.com/Koenkk/zigbee2mqtt/issues/4221

When polling the inverter, the firmware will parse the response. If however that response is not quite correct, strange things are happening :)

[nightmode bug.txt](https://github.com/patience4711/read-APS-inverters-YC600-QS1/files/7625677/nightmode.bug.txt)
[image]
[image]
[image]

If my 'python' poll request is within the same timeframe as the poll request from your firmware, the ESP12 will try to parse a 'corrupt' string and nightmode begins :) 
I've managed to reproduce this error 3 times in a row so there is a high probability that this is the trigger.

No13, 2021-11-30:
When reading the code I spot, what I think is an error in memory allocation:
`Pseudocode:

inMessage[512] = {0}
readZigbee()
  memset( &inMessage, 0, sizeof(inMessage) ) 
          --> sizeof() == ?? How much is cleared here?
  if readCounter < 512:
    processIncomingByte 
          --> read 1 byte from serial; add 2 to inMessage
      readCounter += 1
  ## inMessage can grow to 1024 bytes??
strlen(polling_answer) = 1068 ???`

patience4711, 2021-12-02:
@No13 
Very good work, i overlooked that. This could be the source of other troubles too.  
I changed it to:  
> if (readCounter < CC2530_MAX_SERIAL_BUFFER_SIZE /2) >
This way i don't have to worry about the other use of the readcounter.
Thanks again and keep up the good work.

closed 2021-12-02 by No13