patience4711/read-APSystems-YC600-QS1-DS3 #9: How to handle "AF_DATA_CONFIRM"

adx-git, 2021-12-08:
Debug Output:

pollcmd = 1D2401CFF91414060001000F1380971B01A3D8FBFB06BB000000000000C1FEFE1F
polling answer FE0164010064
no AF_DATA_CONFIRM
polling failed 11
pollcmd = 1D2401CFF91414060001000F1380971B01A3D8FBFB06BB000000000000C1FEFE1F
polling answer FE0164010064FE034480CD14011F
no AF_DATA_CONFIRM
polling failed 11

I tried to understand the code which is printing this error message but failed in understanding what would be expected or is causing this error.

I was able to pair the inverter, so I guess basic communication is working.
How could I debug?

Thanks for your help and btw., great job!
I am following

adx-git, 2021-12-08:
Honestly, I have no idea about the Zigbee protocol but the answer to 2401 as 640100 seems to be correct.

[image]

No13, 2021-12-08:
The second response: FE034480CD14011F

4480 0xCD is NoRoute I believe (ZNwkNoRoute)

Are you out of range?

adx-git, 2021-12-08:
I shouldn't be out of range, since the ESP is very close, less than 1m.
It's actually the same range/position used for pairing.

However, does it mean there is no response from the inverter at all?

patience4711, 2021-12-14:
Yes, no response on polling from the inverter, however it reacts on pairing commands. You can't do anything to debug. You can reboot the inverter, powercycle it (and the esp), sometimes it helps. In my experience this happens on dark rainy days.

adx-git, 2021-12-22:
Thanks!
I would assume if the minimum power for the converted is reached and pairing is working, the polling part should also work, correct?

I am assuming that I am doing something wrong, but what?

closed 2021-12-30 by patience4711