patience4711/read-APSystems-YC600-QS1-DS3 #11: Panel 1 and 2 interchanged

kentbrock, 2022-01-02:
After testing in direct sunlight, I found that panel 1 and 2 are swappedin the overview. Currently, one panel shadows the other very slightly, so I could see that P1 and P2 should be swapped.  This is not a problem for me, because I evaluate the data anyway with the ioBroker. I just noticed

patience4711, 2022-01-03:
So they are not swapped in the mqtt messages?

kentbrock, 2022-01-03:
They are also swapped in the MQTT Massages. As far as I see they are consequently swapped.

patience4711, 2022-01-04:
There was discussion in this subject in the past, wont start that again. Maybe it's a matter of perception on which is panel 0, or you could just swap the plugs of the panels.

closed 2022-01-04 by patience4711

kentbrock, 2022-01-04:
Ok, that was only a notice from me and it isn´t a real problem. The change of the plugs won´t change that because, the inverter inputs are numbered on the inverter enclosure itself. That's the reason I watched it in the first place :-)