patience4711/read-APSystems-YC600-QS1-DS3 #13: datapoint "inverter generated today" suddenly too high

kentbrock, 2022-01-04:
The ESP is running for nearly 6 days without any worth mentioning issues. Because of the weather conditions the typical "inverter generated today" was  about 300Wh. Today there was no sun at all and only around 5 - 20 W per panel - until sunset the datapoint "inverter generated today " was about 4250Wh.  This seems to be far to high. The rise happends between 11:30 (29Wh) to 11:45 (4250Wh).

<html><body>

Time | Type | Command
-- | -- | --
4-16:33:49 | system | start nightmode
4-11:48:16 | mqtt | connected
4-11:38:35 | zigbee | ZB coordinator started
4-11:38:26 | zigbee | zigbee module reset
4-11:38:25 | zigbee | zb down, received :
4-10:18:35 | zigbee | ZB coordinator started
4-10:18:26 | zigbee | zigbee module reset
4-10:18:25 | zigbee | zb down, received :
4-8:35:10 | system | wake-up from nightmode

</body>
</html>

[image]

kentbrock, 2022-01-05:
Hi,

today it happends again - nearly no sun and only low panel power. Up to 11:44 (52Wh) the measurents seemed to be ok. The next value I got was at 11:58 (4718Wh) 

Tomorrow will be a "very" sunny day. Does it make sense to wait if the effect happens only with "no sun" oder should I update the ESP? And should I rise the polling intervall? 

Version: ESP-ECU-v7_01
Pollintervall sec.: 10
system up time: 6 days 6 hrs 11 min.
current errorCode = 11

variables dump
value=0 inverterCount=1 zigbeeUp=1
pollRes=10 now=1641402585 switchonTime=1641371693 switchoffTime=1641400501
unixtime=1641402585
polled = 000000000
resetCounter = 21

<html><body>

5-16:35:1 | system | start nightmode
-- | -- | --
5-13:48:20 | mqtt | connected
5-13:38:39 | zigbee | ZB coordinator started
5-13:38:30 | zigbee | zigbee module reset
5-13:38:29 | zigbee | zb down, received :
5-13:8:20 | mqtt | connected
5-12:58:39 | zigbee | ZB coordinator started
5-12:58:30 | zigbee | zigbee module reset
5-12:58:29 | zigbee | zb down, received :
5-11:58:20 | mqtt | connected
5-11:48:39 | zigbee | ZB coordinator started
5-11:48:30 | zigbee | zigbee module reset
5-11:48:29 | zigbee | zb down, received :
5-10:38:20 | mqtt | connected
5-10:28:39 | zigbee | ZB coordinator started
5-10:28:30 | zigbee | zigbee module reset
5-10:28:29 | zigbee | zb down, received :
5-8:34:54 | system | wake-up from nightmode

</body>
</html>

[image]

patience4711, 2022-01-06:
The most helpfull thing you can do right now is to upgrade the software and keep the high poll rate, so that we can find out if my last surgery solves the problem. Could be that you won't see it for the next 6 days.

I see you have many zigbee resets, this means that the occurence of this high energy value is coincidently (if it's related at all). 
Thanks.

kentbrock, 2022-01-06:
Today everything seems to be normal with firmware version : ESP-ECU-v7_01 - except of Panel 2 which was full of ice today :-( 

I will flash the firmware version : ESP-ECU-v7_02 this evening. 

[image]

<html><body>

6-16:38:22 | mqtt | connected
-- | -- | --
6-16:36:15 | system | start nightmode
6-16:28:42 | zigbee | ZB coordinator started
6-16:28:33 | zigbee | zigbee module reset
6-16:28:32 | zigbee | zb down, received :
6-14:48:42 | zigbee | starting ZB coordinator failed
6-14:48:33 | zigbee | zigbee module reset
6-14:48:32 | zigbee | zb down, received :
6-14:18:42 | zigbee | ZB coordinator started
6-14:18:33 | zigbee | zigbee module reset
6-14:18:32 | zigbee | zb down, received :
6-14:18:24 | mqtt | connected
6-14:8:42 | zigbee | ZB coordinator started
6-14:8:33 | zigbee | zigbee module reset
6-14:8:32 | zigbee | zb down, received :
6-12:58:23 | mqtt | connected
6-12:48:42 | zigbee | ZB coordinator started
6-12:48:33 | zigbee | zigbee module reset
6-12:48:32 | zigbee | zb down, received :
6-12:38:23 | mqtt | connected

</body>
</html>

value=0 inverterCount=1 zigbeeUp=1
pollRes=10 now=1641503355 switchonTime=1641458073 switchoffTime=1641486975
unixtime=1641503355
polled = 100000000
resetCounter = 28

P.S.: I´ve got my RPI Zero today. Is it possible to poll the inverter with the Node MCU and the RPI in parallel? (for sure with a separate CC253x PCB)

patience4711, 2022-01-06:
I would like that too, would give me a lot more testing possibilities.  I tried that with 2 esps without luck. In theory it must be possible i think, you should have 2 coordinators with the same i.d. as you cannot pair to multiple coordinators.

kentbrock, 2022-01-08:
Yesterday it happends again between 13:50 and 14:50 with Version: ESP-ECU-v7_02. 
The Energycounter rises up to 113Wh until 13:50. Suddenly it rises up to 17743Wh. Until 14:50 I got no data points and then it falls down to 136Wh. At the end of the day the energycounter stops at 140Wh. Unfortunately was was not able to check all the data yesterday. So there is no ECU Log :-(

Pollintervall is still 10 seconds.

P.S.: I´ve updated the ECU ota to ESP-ECU-v7_02 but the infopage still shows firmware version : ESP-ECU-v7_01.

patience4711, 2022-01-08:
So i must conclude that there is no direct relation with the zigbee resets. I am still thinking os building in some validity check for the energy values but is not so easy to catch them all. Anyway the best thing to do is lower the polling rate, no use to poll that often. The gaps in your graphs will dissapear then. Like i said before, polling once at 5 minutes gives the best stability.

PS yes i forgot to update the version nr.

kentbrock, 2022-01-10:
With the 5 minutes polling rate it´s far better! I assume that the calculation of kWh happens in the inverter not on the ESP?

patience4711, 2022-01-12:
Correct. The inverter provides the amount of energy produced. From that the ESP calculates the power. There is a thing to notice: During twiligth the inverter can "wackel" around the minimum power it needs to work, causing restarts. At restart the energy is always zero. This means that we produce a little more power than reported..... 

I even suspect that when the power gets too low during a zigbee transmission, the message can be damaged, causing a reset of the zigbee module. In the ESP you can set an offset relative to sunrise (i use 10 minutes) which minimizes the occurence of this phenomenon.

kentbrock, 2022-01-21:
Its runnig pretty stable with the 5 minutes poll rate. No issues anymore for 13 days. From my point of view this issue can be closed.

closed 2022-01-25 by patience4711