patience4711/read-APSystems-YC600-QS1-DS3 #14: polling failed 50

ZnarfMe, 2022-01-07:
Hi @patience4711 

great work. After pairing my four QS1 devices I am getting the data to the MQTT Host.

Time by Time the device is getting stuck at the "polling failed 50" issue.

As an workaround I am sending the init "10;INIT_N" command and then it is working again for round round about a hour. 

Is there any chance to send the command automatically as an workaround? Any chance to that I can help you investigating the root cause? 

`* * * * polled inverter 1 * * * *
panel 0 power 9.00 energy today 37.77
panel 1 power 6.93 energy today 19.71
panel 2 power 2.08 energy today 6.89
panel 3 power 18.01 energy today 52.57
total inverter: pwr 36.01 en: 116.94
polling answer FE0xxxE1D
the timespan = 11
* * * * polled inverter 2 * * * *
panel 0 power 103.51 energy today 303.82
panel 1 power 108.80 energy today 312.55
panel 2 power 101.24 energy today 305.17
panel 3 power 98.22 energy today 317.44
total inverter: pwr 411.77 en: 1238.97
polling answer FE0xxxE0E
the timespan = 11
* * * * polled inverter 3 * * * *
panel 0 power 21.16 energy today 67.65
panel 1 power 25.69 energy today 95.84
panel 2 power 23.42 energy today 68.15
panel 3 power 7.56 energy today 76.59
total inverter: pwr 77.82 en: 308.24
...
polling answer
polling failed 50
zb send poll cmd inverter 0
polling answer
polling failed 50
zb send poll cmd inverter 1
polling answer
polling failed 50
zb send poll cmd inverter 2
polling answer
polling failed 50
zb send poll cmd inverter 3
polling answer
polling failed 50
...
command = 10;INIT_N
init zb coordinator
zigbee module reset
zb send cmd 0
inMessage = FE0xxx062 rc = 17
zb send cmd 1
...
* * * polled inverter 0 * * * *
panel 0 power 94.75 energy today 343.29
panel 1 power 241.57 energy today 516.72
panel 2 power 246.56 energy today 574.37
panel 3 power 144.61 energy today 494.74
total inverter: pwr 727.49 en: 1929.13
* * * * polled inverter 1 * * * *
panel 0 power 9.97 energy today 38.09
panel 1 power 7.20 energy today 19.96
panel 2 power 2.77 energy today 6.97
panel 3 power 21.61 energy today 53.28
total inverter: pwr 41.56 en: 118.30
polling answer FE0xxxE60
the timespan = 15
* * * * polled inverter 2 * * * *
panel 0 power 101.95 energy today 307.17
panel 1 power 106.93 energy today 316.06
panel 2 power 99.73 energy today 308.45
panel 3 power 96.41 energy today 320.62
total inverter: pwr 405.02 en: 1252.29
polling answer FE0xxx026
the timespan = 15
* * * * polled inverter 3 * * * *
panel 0 power 21.05 energy today 68.35
panel 1 power 26.04 energy today 96.70
panel 2 power 23.82 energy today 68.92
panel 3 power 7.76 energy today 76.83
total inverter: pwr 78.68 en: 310.80
...
polling answer
polling failed 50
zb send poll cmd inverter 2
...`

patience4711, 2022-01-07:
@ZnarfMe 
The command is send automatically when the healtcheck proves that the zigbee is down. (every 10 minutes).
The root cause of these problems is a high polling rate. When you poll every 5 minutes, the problem is gone. The official ECU polls every 5 min.

ZnarfMe, 2022-01-23:
Hi @patience4711,

it took me a while for investigating the issue in more detail. My two neighbors do also use APSystems. They both having an ECU-B device for monitoring. If they shutdown their devices it works without any problem. If they also running I have the pairing issue. Even the 10 minute healthcheck does not work for me in this case. Any chance to get a force direct repairing setting?

Thank You

patience4711, 2022-01-25:
That cannot be the solution, what i would do is pair the inverters when the neighbours have their devices shutted-off. Once paired, there is no need for re-pairing. If you won't get polling answers this must have another cause. Try with polling rate 5 minutes. And test with the esp closer to the inverter.
I went to the Raspberry so there is no development in the software for the esp anymore. I don't have means to test. Sorry.

closed 2022-01-31 by patience4711