patience4711/read-APSystems-YC600-QS1-DS3 #17: TESTING VERSION 8_2

patience4711, 2022-02-21:
This issue is opend for the testing process of the new version.

manoficons, 2022-02-21:
Here is the polling result from console (v8_0)

zb send poll cmd inverter 0
--
polling  answer  FE0164010064FE034480001401D2FE0345C4570700D2FE7244810000060157071414000F004ABBBB00005E408000191535FBFB51B103C20F44690000340000000005816C0A616F04D70000000000000000C6A6973000CAE5150F1F00030553073F0303030100000100000000000000000000000000000000000000000000000000000000000000FEFE57070ED7
the timespan = 53
* * * * polled inverter 0 * * * *
panel 0  power 7974.48  energy increase 117.40
panel 1  power 8144.94  energy increase 119.91
no panel 2
no panel 3
zb send poll cmd inverter 1
polling  answer  FE0164010064FE034480001401D2FE0345C4364400F0FE72448100000601364414140046001EC2BB00005E408000038812FBFB51B103FA0F441D0000314000000081907183407104D90000000000000000626BA01B0061447A081F00030556073F0303030100000100000000000000000000000000000000000000000000000000000000000000FEFE36440E38
the timespan = 20
* * * * polled inverter 1 * * * *
panel 0  power 10469.78  energy increase 58.17
panel 1  power 10347.19  energy increase 57.48
no panel 2
no panel 3

manoficons, 2022-02-21:
Btw. this wasn't a manual poll. I had the console open during normal system polling interval if it makes a difference.

patience4711, 2022-02-21:
HI, thanks. This is very helpfull. The calculations are correct but there is something wrong with the timespan. That should be 300 since this version polles once in 5 minutes, hard coded. 
Working on it...

manoficons, 2022-02-23:
In v8_0_BETA2 the values look good! But it seems like the console cannot be used. I can type commands but nothing happens.
Btw. I tested manual polling over http. This seems to work. Great feature!

patience4711, 2022-02-23:
Oke thanks. I'l wait a few days to collect any other issues and than release it. In the mean time, if you have the opportunity to post another poll result here, would be great. Prefeably two consutive polls.
Thanks in advance

manoficons, 2022-02-23:
Is it possible to get the raw poll results without a working console?

patience4711, 2022-02-23:
Archhhh, it sometimes drives me crazy. Anyway issue 1 is solved.

https://1drv.ms/u/s!AkMG5FvUwhedixojxJB82etIhtnr?e=yCi5w9

manoficons, 2022-02-23:
Sorry, I still have no input or output on the console in Firefox. In Opera Browser opening the console I get
[image]
:

patience4711, 2022-02-23:
I had that too so i copied the console from version7 over and than it worked.
Do you have "ESP-ECU-v8_0_BETA3" now? Could be something else went wrong..

patience4711, 2022-02-23:
Thanks, 
Yes these values look allright.  I still see strange values for the timespan. Thats probably because the poll is manual. Can you please confirm that when you look in the console that the automatic polls have a timespan of round 300? 
And how about the total energy at the end of the day, does that seem to be correct?

manoficons, 2022-02-23:
For today the total energy values (inverter 0 and 1) seem ok.
Regarding the polling interval I will check the console tomorrow.

manoficons, 2022-02-25:
Was too late yesterday to get some readings. These are my first polling values today. There is still snow on the panels...

ping ok

inMessage = FE0E670000FFFF80971B01A3D8000007090011 rc = 19
zb send poll cmd inverter 0
polling  answer  FE0164010064FE034480001401D2FE0345C4570700D2FE7244810000060157071414002200E94DC600005E408000191535FBFB51B103B40F424B000013320000002AD06B2A306B04C80000000000000000097E3D92000A48F24D1F00030554073F0303030100000100000000000000000000000000000000000000000000000000000000000000FEFE57070E56
* * * * polled inverter 0 * * * *
extracted time = 4914  the timespan = 300
* * * decoding panel 0 * * *
en_incr 0.63  power 7.56
* * * decoding panel 1 * * *
en_incr 0.65  power 7.78
no panel 2
no panel 3
zb send poll cmd inverter 1
polling  answer  FE0164010064FE034480001401D2FE0345C4364400F0FE7244810000060136441414004C00255CC600005E408000038812FBFB51B103EE0F402F00000D7400000001A07B00707B04CE00000000002800000004EFCB0000637ADD1F0003015C073F0303030100000100000000000000000000000000000000000000000000000000000000000000FEFE36440EF3
* * * * polled inverter 1 * * * *
extracted time = 3444  the timespan = 300
* * * decoding panel 0 * * *
en_incr 0.00  power 0.00
* * * decoding panel 1 * * *
en_incr 0.02  power 0.25
no panel 2
no panel 3

Looking at the total values of the day (yesterday evening) I found that only inverter 0 showed the summarized energy, inverter 1 didn't.

manoficons, 2022-02-25:
One more  automatic poll

ping ok
inMessage = FE0E670000FFFF80971B01A3D8000007090011 rc = 19
zb send poll cmd inverter 0
polling  answer  FE0164010064FE034480001401D2FE0345C4570700D2FE7244810000060157071414001F004B32AC00005E408000191535FBFB51B103BB0F4006000025F3000000FE107202917304DA00000000000000003D050DC5003FE911C81F00030558073F0303030100000100000000000000000000000000000000000000000000000000000000000000FEFE57070E33
* * * * polled inverter 0 * * * *
extracted time = 9715  the timespan = 300
* * * decoding panel 0 * * *
en_incr 4.05  power 48.56
* * * decoding panel 1 * * *
en_incr 4.19  power 50.34
no panel 2
no panel 3
zb send poll cmd inverter 1
polling  answer  FE0164010064FE034480001401D2FE0345C4364400F0FE72448100000601364414140046009340AC00005E408000038812FBFB51B103F80F3F830000203400000061707A58707804E60000000000000000135711CA001539804B1F00030553073F0303030100000100000000000000000000000000000000000000000000000000000000000000FEFE36440EEC
* * * * polled inverter 1 * * * *
extracted time = 8244  the timespan = 300
* * * decoding panel 0 * * *
en_incr 1.42  power 17.04
* * * decoding panel 1 * * *
en_incr 1.59  power 19.06
no panel 2
no panel 3

patience4711, 2022-02-25:
Well the polling and calculations look good.  

The total energy values shown in the frontpage you mean?  Can you make a screenshot? Is this consequently during the day? And consequently for the same inverter?  Or does it happen only after sunset?

manoficons, 2022-02-25:
During the day everything seems ok. Both vlaues (I have two yc600 inverters) are summarized as they should. I think it happens only after sunset. Then inverter 1 shows no longer the value. I will have al look later.

manoficons, 2022-02-25:
Ok, it looks like both inverters show the same - no values after sunset.
[image]
[image]

patience4711, 2022-02-26:
Oke, i think i solved that too. The value should now stay untill midnight.

I changed something else: In the Raspberry version i noticed that the inverter already starts to work 10 minutes before sunrise (my panels are layed horizontal) so i changed the selectbox for offset / sunset/rise such that there is a negative value possible. Starting too late with the polling is harmless but stopping too early means lost of data!  Now you can start 15 min. before sunrise and stop 15 min. after sunset, if necessary.

In the past, starting too early caused several in vain pollings which in turn caused reset of the zigbee module. Hopefully with the low polling rate this doesn't happen anymore.

BETA5 https://1drv.ms/u/s!AkMG5FvUwhediyChaLtx_RjeJvbL?e=DXkMhE

patience4711, 2022-03-04:
I assume you mean the value 'inverter generated today'. This value is only zero-ed at midnight or at a reboot. 
Oh yes, and at a zigbee  failure. This is needed to get a nice clean restart of the system. 

But for the total energy this is not necessary so i changed that now. 
BETA 6 https://1drv.ms/u/s!AkMG5FvUwhediyMPDEqaXDXNIjcd?e=RxCzrG

manoficons, 2022-03-07:
With BETA6 I got only "checking / initialyzing the zigbee module..." and no output.  Is it necessary to wipe files or clear esp chip before upgrade?
Back to BETA5 output works fine.

patience4711, 2022-03-07:
No, there is only a very small difference between the two. Are you sure you downloaded it from the above link? There is a test version that doesn't start the zigbee system. I know someone downloaded from the link above and it works.  So if you would try it again it should work.

manoficons, 2022-03-28:
System working now for some days with BETA6.

firmware version : ESP-ECU-v8_0_BETA6
time retrieved today : yes
the signalstrength of the connection to the router = -68
mosquitto not configured
check the mosquitto settings
system up time: 19 days 23 hrs 20 min.
current errorCode = 0
polling starts at 6:58 and stops at 19:40

I see some errors in the log from time to time like "zb down, zb module reset, starting ZB coordinator failed".

All values sent seem ok. I only use http no mosquitto.

closed 2022-03-28 by patience4711