patience4711/read-APSystems-YC600-QS1-DS3 #19: TESTING DS3

patience4711, 2022-02-24:
There is a first testversion ready for the integration with the new APS inverter DS3.
With this version we should be able to pair the DS3. 
Download ESP-ECU-DS3-v1_0: https://1drv.ms/u/s!AkMG5FvUwhedix7XH5n9z9QSZAHa?e=RELkgf
This version has many debug possibilities sothat we can follow the zigbee communication. 

In order to test you can follow these steps.
- in settings putoff the automatic polling
- define an inverter0 in the menu inverters
- go to the console and type 10;diag
- type 10;INIT_P check that the coordinator has been started, if not, check hardware/wiring
- when coordinator ok, open a new webpage at the ecu and go to the menu inverters
- if you click 'pair' for your inverter, you can follow the proces in the other page where the console is open.
- copy the output of the console and post it here.

If your inverter was already paired with the ECU-b, it may be necessary to power-cycle the inverter.

jenswalit, 2022-02-25:
Hello, my test`

init zb coordinator
--
zigbee module reset
zb send cmd 0
inMessage = FE064180010202020702C1FE0166050062 rc = 17
zb send cmd 1
inMessage = FE064180020202020702C2 rc = 11
zb send cmd 2
inMessage = FE0166050062 rc = 6
zb send cmd 3
inMessage = FE0166050062 rc = 6
zb send cmd 4
inMessage = FE0166050062 rc = 6
zb send cmd 5
inMessage = FE0166050062 rc = 6
zb send cmd 6
inMessage = FE0164000065 rc = 6
zb send cmd 7
inMessage = FE00660066FE0145C0088CFE0145C0088C rc = 17
inMessage = FE0E670000FFFF971B01A3D800000007080090FE0145C0088CFE0145C0088C rc = 31
inMessage = FE0E670000FFFF971B01A3D800000007080090FE0145C0098D rc = 25
inMessage = FE0E670000FFFF971B01A3D800000007090091 rc = 19
starting ZB coordinator failed

</body>
</html>`

iLeopod, 2022-02-27:
Also have an DS3 to test:
`

poll inverter 0
--
zb send poll cmd inverter 0
polling answer FE0164010064FE034480001401D2FE274481000006000000141400FF0000000000001380971B01A3D8FBFB06BB000000000000C1FEFE00000002
no route receipt
ignoring, received FE0164010064FE034480001401D2FE274481000006000000141400FF0000000000001380971B01A3D8FBFB06BB000000000000C1FEFE00000002
polling failed 15

`

patience4712, 2022-02-28:
Hi @iLeopod 
Thanks for your interesting contribution.  At first glance this looks like a good poll answer. This would mean that you have succeeded  to pair the inverter, can you confim that?  
With the ESP-ECU-DS3?

patience4712, 2022-02-28:
Hi @jenswalit 
This polling attempt failed, this is sometimes difficult when the inverter was paired already. Could be that you have to change the ECU id slightly. Anyway, so it seems we can pair the DS3, so thats alread y good step forward.

iLeopod, 2022-02-28:
incomplete log of pairing(maybe is already paird?):
```
inMessage = FE0164000065 rc = 6
zb send cmd 7
inMessage = FE00660066FE0145C0098D rc = 11
try pairing inv 0
pair command = 2524020FFFFFFFFFFFFFFFFF14FFFF140D0200000F1100703000019291FFFF10FFFF80971B01A3D839
sending paircmd 2
received : FE0164020067FE034480001400D3
pair command = 1A24020FFFFFFFFFFFFFFFFF14FFFF140C0201000F060070300001929177
sending paircmd 3
received : FE0164020067FE034480001401D2
not found serialnr, returning
pair command = 2524020FFFFFFFFFFFFFFFFF14FFFF140F0102000F1100703000019291A3D810FFFF80971B01A3D841
sending paircmd 4
received : FE0164020067FE034480001402D1
not found serialnr, returning
pair command = 1A24020FFFFFFFFFFFFFFFFF14FFFF14010103000F060080971B01A3D84F
sending paircmd 5
received : FE0164020067FE034480001403D0
failed, inverter got id 0x0000
sending NO cmd
inMessage = FE0164010064FE034480001401D2 rc = 14
inMessage = FE0E670000FFFF80971B01A3D8000007090011 rc = 19

```

patience4711, 2022-03-01:
@iLeopod  @jenswalit 
Thanks for testing, i didn't see any successfull pairing so i can't make much out of it at the moment. With these pairing commands it should however be possible to pair, someone did it already. 

@iLeopod I thought i saw a good pollinganswer but that wasn't the case. 
Already paired shouldn't make a difference here, unless it is paired to another ecu-id that is still in the air.

What is strange; the software doesn't poll an inverter that is not paired, it needs have a short-id, which it gets at successfull pairing. Now another pairing attempt failed you can't poll it anymore i think.

jenswalit, 2022-03-01:
@patience4711 with Z-Tool i can paired. 

`Tue, 1 Mar 2022 06:50:08 UTC: Setup completed for 0xD8A3011B9780, starting application...
Tue, 1 Mar 2022 06:50:15 UTC: Application has started. Pairing with 703000021300
Tue, 1 Mar 2022 06:50:16 UTC: Starting pairing, phase 1
Tue, 1 Mar 2022 06:50:16 UTC: Sent: APS#=0, clusterID=0x20d, Data=0x70;0x30;0x00;0x02;0x13;0x00;0xFF;0xFF;0x10;0xFF;0xFF;0x80;0x97;0x1B;0x01;0xA3;0xD8;
Tue, 1 Mar 2022 06:50:20 UTC: Starting pairing, phase 2
Tue, 1 Mar 2022 06:50:20 UTC: Sent: APS#=1, clusterID=0x20c, Data=0x70;0x30;0x00;0x02;0x13;0x00;
Tue, 1 Mar 2022 06:50:21 UTC: Received message from inverter 0x6c22 with ClusterID: 0x101; Data: ff1a703000021300
Tue, 1 Mar 2022 06:50:21 UTC: Starting pairing, phase 3
Tue, 1 Mar 2022 06:50:21 UTC: Sent: APS#=2, clusterID=0x10f, Data=0x70;0x30;0x00;0x02;0x13;0x00;0xA3;0xD8;0x10;0xFF;0xFF;0x80;0x97;0x1B;0x01;0xA3;0xD8;
Tue, 1 Mar 2022 06:50:22 UTC: Received message from inverter 0x6c22 with ClusterID: 0x101; Data: ffff703000021300
Tue, 1 Mar 2022 06:50:23 UTC: Starting pairing, phase 4
Tue, 1 Mar 2022 06:50:23 UTC: Sent: APS#=3, clusterID=0x101, Data=0x80;0x97;0x1B;0x01;0xA3;0xD8;
Tue, 1 Mar 2022 06:50:25 UTC: Completed.
Tue, 1 Mar 2022 06:50:48 UTC: Device is being configured.
Tue, 1 Mar 2022 06:50:48 UTC: Wrote configuration string to NVR: 3
Tue, 1 Mar 2022 06:50:48 UTC: Wrote configuration string to NVR: 3
Tue, 1 Mar 2022 06:50:49 UTC: Wrote configuration string to NVR: 255,255,128,151,27,1,163,216
Tue, 1 Mar 2022 06:50:49 UTC: Wrote configuration string to NVR: 255,255,128,151,27,1,163,216
Tue, 1 Mar 2022 06:50:50 UTC: Wrote configuration string to NVR: 0
Tue, 1 Mar 2022 06:50:50 UTC: Wrote configuration string to NVR: 216,163
Tue, 1 Mar 2022 06:50:50 UTC: Wrote configuration string to NVR: 0,0,1,0
Tue, 1 Mar 2022 06:50:50 UTC: Wrote configuration string to NVR: 1
Tue, 1 Mar 2022 06:50:51 UTC: Setup completed for 0xD8A3011B9780, starting application...
Tue, 1 Mar 2022 06:50:58 UTC: Application has started.
Tue, 1 Mar 2022 06:50:58 UTC: Waiting some time so we can discover what is around.
Tue, 1 Mar 2022 06:51:58 UTC: Sending broadcast to all the inverters (Honey, I'm back ;-).
Tue, 1 Mar 2022 06:52:09 UTC: Sending initialization command to 0x6c22 - 703000021300
Tue, 1 Mar 2022 06:52:09 UTC: Received message from 703000021300 with ClusterID: 0x106; Data: 703000021300fbfb5cbbbb2000fc0001ffff000000000000a000067a063e000d000e03681383036b00040020ffff04dd051f000085fb00007e2c00ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff3702fefe
Tue, 1 Mar 2022 06:52:12 UTC: Waiting for Data...
Tue, 1 Mar 2022 06:53:09 UTC: ID: 703000021300 Fq: 775.050[Hz] T: 12924.5[C] ACV: 303.1[V] P1: 0.0[W] P2: 2.2[W] E1: 0.02[kWh] E2: 20.73[kWh]  DC1: 0.0[A] DV1: 0.0[V] DC2: 9.6[A] DV2: 36.7[V] Raw data: 703000021300fbfb5cbbbb2000fc0001ffff0000000000000000066e0652000f000e0366138903a700050021ffff04ed052800009a2f0000920d00ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff3605fefe
Tue, 1 Mar 2022 06:54:09 UTC: ID: 703000021300 Fq: 775.050[Hz] T: 12924.5[C] ACV: 305.7[V] P1: NaN[W] P2: -Infinity[W] E1: 0.02[kWh] E2: 20.43[kWh]  DC1: 0.0[A] DV1: 0.0[V] DC2: 9.6[A] DV2: 29.0[V] Raw data: 703000021300fbfb5cbbbb2000fc0001ffff000000000000000006570660000f000e0367138703e300040021ffff04fe052f0000aff30000a6fb00ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff3829fefe
`

but with cc2530 an your ESP Version come this.

`inMessage = FE00660066FE0145C0088CFE0145C0088C rc = 17
try pairing inv 0
pair command = 2524020FFFFFFFFFFFFFFFFF14FFFF140D0200000F1100703000021300FFFF10FFFF971B01A4D90xAC
sending paircmd 2
received : FE0145C0088CFE0164020067FE0145C0088CFE0145C0088CFE0145C0098D
pair command = 1A24020FFFFFFFFFFFFFFFFF14FFFF140C0201000F060070300002130064
sending paircmd 3
received : FE0164020067
not found serialnr, returning
pair command = 2524020FFFFFFFFFFFFFFFFF14FFFF140F0102000F1100703000021300D90x10FFFF971B01A4D90x76
sending paircmd 4
received : FE0164020067
not found serialnr, returning
pair command = 1A24020FFFFFFFFFFFFFFFFF14FFFF14010103000F0600971B01A4D90xC9
sending paircmd 5
received : FE0164020067
failed, inverter got id 0x0000
sending NO cmd
inMessage = FE0164010064FE034480001401D2 rc = 14
inMessage = FE0E670000FFFF971B01A4D900000007090097 rc = 19
inMessage = FE0E670000FFFF971B01A4D900000007090097 rc = 19
inMessage = FE0E670000FFFF971B01A4D900000007090097 rc = 19`

jenswalit, 2022-03-01:
@patience4711 can you upload the DS3 source. So i can testing

patience4711, 2022-03-01:
Hi Jens here is the ino
https://1drv.ms/u/s!AkMG5FvUwhediyEDg9-ijPOyYurR?e=0YDIaE
You will see that the paring commands are, apart from the prefixes, equal to the Z-tool commands. So i think it should work.

I remember that pairing failed mostly when the inverter was already paired, waiting some time or powercycle the inverter helped.

jenswalit, 2022-03-01:
That's right, after a while you connected it

`<html>
<body>

inMessage = FE0164000065 rc = 6
--
zb send cmd 7
inMessage = FE00660066FE0145C0088CFE0145C0088C rc = 17
try pairing inv 0
pair command = 2524020FFFFFFFFFFFFFFFFF14FFFF140D0200000F1100703000021300FFFF10FFFF971B01A4D90xAC
sending paircmd 2
received : FE0145C0088CFE0164020067FE0145C0088CFE0145C0088CFE0145C0098D
pair command = 1A24020FFFFFFFFFFFFFFFFF14FFFF140C0201000F060070300002130064
sending paircmd 3
received : FE0164020067FE1A448100000C0200001414018300FB5900000006703000021300226C0DE5FE0345C4226C00CCFE1C448100000101226C1414008300C75E00000008FF1A703000021300226C0E71
pair command = 2524020FFFFFFFFFFFFFFFFF14FFFF140F0102000F1100703000021300D90x10FFFF971B01A4D90x76
sending paircmd 4
received : FE0164020067FE25448100000F0100001414018300CA6C00000011703000021300D90010FFFF971B01A4D900226C0DF0FE0345C4226C00CCFE1C448100000101226C1414008000927300000008FFFF703000021300226C0EEF
pair command = 1A24020FFFFFFFFFFFFFFFFF14FFFF14010103000F0600971B01A4D90xC9
sending paircmd 5
received : FE0164020067FE1A44810000010100001414018300837F00000006971B01A4D900226C0D14
success, inverter got id 0x6C22
sending NO cmd
inMessage = FE0164010064FE034480001401D2 rc = 14
inMessage = FE0E670000FFFF971B01A4D900000007090097 rc = 19
inMessage = FE0E670000FFFF971B01A4D900000007090097 rc = 19
inMessage = FE0E670000FFFF971B01A4D900000007090097 rc = 19

</body>
</html>`

patience4711, 2022-03-01:
Oke so far so good, now you could try a poll to see if there is any answer.

jenswalit, 2022-03-01:
@patience4711  I had changed for testing. So here. But not  971B01A4D90xAC
[image]

patience4711, 2022-03-02:
OKE, the ecu id is used inverted. You can see that looking at the commands.

So now we can safely assume that the pairing works. I am already so far that i can decode a poll answer. So if the assumptions of npeter are correct, the only thing i have to know is how to trigger  poll and then we are done.

jenswalit, 2022-03-07:
@patience4711 Ok, what should I send to the inverter. do you have an example

patience4711, 2022-03-07:
Hi @jenswalit 
It's easy, let's assume that your inverternr = 0. Be sure it is paired.
Just go to the console, type **10;diag**, if you see set diagnose to 1, oke. Else type **10;diag** again.
Now you type **10;poll=0** and you will see the magic. 

In the poll command you can recognize inverter's short id 6c33 and your ecu id inversed i think.

If you click help, you'l see what more you can do with the console. 

What i am interested in is can we get get some answer from the inverter.

jenswalit, 2022-03-07:
@patience4711 

`<html>
<body>

try pairing inv 0
--
pair command = 2524020FFFFFFFFFFFFFFFFF14FFFF140D0200000F1100703000021300FFFF10FFFF80971B01A3D82A
sending paircmd 2
received : FE0145C0088CFE0164020067FE0145C0088CFE0145C0088CFE0145C0088CFE0145C0098D
pair command = 1A24020FFFFFFFFFFFFFFFFF14FFFF140C0201000F060070300002130064
sending paircmd 3
received : FE0164020067FE1A448100000C0200001414014E00DC6000000006703000021300226C0D36FE0345C4226C00CCFE1C448100000101226C1414004C00226100000008FF1A703000021300226C0E64
pair command = 2524020FFFFFFFFFFFFFFFFF14FFFF140F0102000F1100703000021300A3D810FFFF80971B01A3D852
sending paircmd 4
received : FE0164020067FE25448100000F0100001414014C00A37300000011703000021300A3D810FFFF80971B01A3D8226C0D6DFE0345C4226C00CCFE1C448100000101226C1414004C002A7600000008FFFF703000021300226C0E9E
pair command = 1A24020FFFFFFFFFFFFFFFFF14FFFF14010103000F060080971B01A3D84F
sending paircmd 5
received : FE0164020067FE1A44810000010100001414014C0058860000000680971B01A3D8226C0D7F
success, inverter got id 0x6C22
sending NO cmd
inMessage = FE0164010064FE034480001401D2 rc = 14
inMessage = FE0E670000FFFF80971B01A3D8000007090011 rc = 19
poll inverter 0
zb send poll cmd inverter 0
polling answer FE0164010064FE034480001401D2FE0345C4226C00CCFE0345C4226C00CC
no AF_INCOMING_MSG
polling failed 13

</body>
</html>`

patience4711, 2022-03-07:
@jenswalit Thanks, so that doesn't work. I understand that you are going to sniff again to verify the values of npeter?  I am interested in the string that is send from ecu to the inverter preceeding a poll answer. With that we can find out to trigger the poll.

MarvinGS, 2022-03-08:
I just received my DS3-S today. Do i need an original ECU-B to support this project, too? Or is an ESP with a cc2530 enough?

lumbsta, 2022-03-09:
Hi,

I successfully paired my DS3 inverter and tried to poll it with 10;poll=0. This is what I get:
> * * connection opened * *
set diagnose to 1
poll inverter 0
zb send poll cmd inverter 0
polling answer FE0164010064FE034480001401D2FE0345C4C951001AFE0345C4C951001A
no AF_INCOMING_MSG
polling failed 13

jenswalit, 2022-03-09:
@patience4711 here is another trace.
[Trace080322.log](https://github.com/patience4711/read-APSystems-YC600-QS1/files/8213734/Trace080322.log)

patience4711, 2022-03-09:
@lumbsta @MarvinGS 
We are still investigating, It looks like we will succeed to make the esp implementation ( and raspberry)  suittable for the ds3. In that case you won't need an ECU.

patience4711, 2022-03-09:
@jenswalit  @lumbsta 
I made a new testversion with a slightly different poll command. Could you update your esp to this version (use fwupdate)   and try to pair and poll your ds3? If you check polling in the settings page, the polling is every 5 minutes so i would say let it work for some time.
I am interested in the poll command which will be logged in the console.

https://1drv.ms/u/s!AkMG5FvUwhediyUfRF7wxqNwk8k0?e=kQHnqH

lumbsta, 2022-03-09:
@patience4711 
Here you go:

> * * connection opened * *
set diagnose to 1
zb send poll cmd inverter 0 cmd:1D2401C9511414000601000F1380971B01A3D8FBFB06BB000000000000C1FEFEB1
polling answer FE0164010064FE034480001401D2
no route receipt
ignoring, received FE0164010064FE034480001401D2
polling failed 15
zb send poll cmd inverter 0 cmd:1D2401C9511414000601000F1380971B01A3D8FBFB06BB000000000000C1FEFEB1
polling answer FE0164010064FE034480001401D2
no route receipt
ignoring, received FE0164010064FE034480001401D2
polling failed 15
inMessage = FE02610179071C rc = 7
ping ok
inMessage = FE0E670000FFFF80971B01A3D8000007090011 rc = 19

lumbsta, 2022-03-09:
Here's my polling answer:
> zb send poll cmd inverter 0 
cmd:1D2401C9511414060001000F1380971B01A3D8FBFB06BB000000000000C1FEFEB1
polling answer FE0164010064FE034480001401D2FE0345C4C951001AFE0345C4C951001A
no AF_INCOMING_MSG
polling failed 13

patience4711, 2022-03-09:
@lumbsta Thankzz The last 2 answers contains your inverter's short-id C951. In the inverters page you should see 0x51C9.
So i think that looks good. 
If you leave it working tonight we'l see if the inverter reset makes a difference. 
It wil help that the esp is not too far away from the inverter but that's obvious. Within a range of 4 mtr it surely should work.

jenswalit, 2022-03-10:
@patience4711  I have something
`<html>
<body>

wake-up from nightmode
--
reset power values
reset energy values
polling failed 10
polling failed 10
polling failed 10
polling failed 10
polling failed 10
polling failed 10
polling failed 10
polling failed 10

</body>
</html>`

lumbsta, 2022-03-10:
@patience4711 
Same output as seen before, now in a loop:

> zb down, received :
reset power values
init zb coordinator
zigbee module reset
zb send cmd 0
inMessage = FE064180010202020702C1FE0166050062 rc = 17
zb send cmd 1
inMessage = FE064180020202020702C2 rc = 11
zb send cmd 2
inMessage = FE0166050062 rc = 6
zb send cmd 3
inMessage = FE0166050062 rc = 6
zb send cmd 4
inMessage = FE0166050062 rc = 6
zb send cmd 5
inMessage = FE0166050062 rc = 6
zb send cmd 6
inMessage = FE0164000065 rc = 6
zb send cmd 7
inMessage = FE00660066FE0145C0088C rc = 11
sending NO cmd
inMessage = FE0145C0098DFE0164010064FE034480001401D2 rc = 20
inMessage = FE0E670000FFFF80971B01A3D8000007090011 rc = 19
ZB coordinator started
zb send poll cmd inverter 0 cmd:1D2401C9511414060001000F1380971B01A3D8FBFB06BB000000000000C1FEFEB1
polling answer FE0164010064FE034480001401D2FE0345C4C951001AFE0345C4C951001A
no AF_INCOMING_MSG
polling failed 13

...and so on.

jenswalit, 2022-03-10:
@patience4711 
`inMessage = FE0E670000FFFF80971B01A3D8000007090011 rc = 19
inMessage = FE02610179071C rc = 7
ping ok
inMessage = FE0E670000FFFF80971B01A3D8000007090011 rc = 19
inMessage = FE02610179071C rc = 7
ping ok
inMessage = FE0E670000FFFF80971B01A3D8000007090011 rc = 19
inMessage = FE02610179071C rc = 7
ping ok
inMessage = FE0E670000FFFF80971B01A3D8000007090011 rc = 19
inMessage = FE02610179071C rc = 7
ping ok
inMessage = FE0E670000FFFF80971B01A3D8000007090011 rc = 19
inMessage = FE02610179071C rc = 7
ping ok
inMessage = FE0E670000FFFF80971B01A3D8000007090011 rc = 19
inMessage = FE02610179071C rc = 7
ping ok
inMessage = FE0E670000FFFF80971B01A3D8000007090011 rc = 19
inMessage = FE02610179071C rc = 7
ping ok
inMessage = FE0E670000FFFF80971B01A3D8000007090011 rc = 19`

patience4711, 2022-03-10:
@jenswalit  Yours is not polling, what you see is only the result of a healthcheck. In 'settings' you should check 'polling. 
@lumbsta 
So it does not seem to work. To rule out that it is the software we can do another test with the present version for the yc600/qs1, we know that this works. Since the polling commands are the same we could try this to see if there are polling answers.

https://1drv.ms/u/s!AkMG5FvUwhediyTg0tSR5dSClawQ?e=JtjgD7

If you installed this version you should set your DS3 inverter as a yc600. 

If that doesn't work either, has anyone of you some experience with running a lua script on an Esp?.

lumbsta, 2022-03-10:
@patience4711 

> set diagnose to 1
--
zb send poll cmd inverter 0
polling answer FE0164010064FE034480001401D2FE0345C4C951001AFE0345C4C951001A
no AF_INCOMING_MSG
polling failed 13

Again, same output as before.

Haven't run any .lua scripts yet, but I found a bunch of tutorials.

patience4711, 2022-03-10:
Oke Thanks. I am really puzzled now. With z-tool we send a command and get a response, sending the same command with our own program and it doesn't work. 
We could give it a try with the lua implementation of kadszol, that has a slightly different approach. I made a package that contains all you need and a readme.
https://1drv.ms/u/s!AkMG5FvUwhediyczfjIomMdb5rGz?e=ktFUxf

lumbsta, 2022-03-11:
@patience4711 
I just ordered a CC2531 to do the sniffing with Z-Tool, should arrive tomorrow.
I really appreciate your work, so I want to contribute as much as needed to make the DS3 work!

patience4711, 2022-03-11:
@lumbsta Well that's great, thanks. 
You followed the discussion on https://github.com/Koenkk/zigbee2mqtt/issues/4221 so you know what and how to test. Not something i can help with. @jenswalit has experience with this sniffing work, he can tell you how to setup the query with z-tool.

patience4711, 2022-03-12:
@iLeopod 
Some time ago you posted this:

> zb send poll cmd inverter 0
polling answer FE0164010064FE034480001401D2FE274481000006000000141400FF0000000000001380971B01A3D8FBFB06BB000000000000C1FEFE00000002
no route receipt
ignoring, received FE0164010064FE034480001401D2FE274481000006000000141400FF0000000000001380971B01A3D8FBFB06BB000000000000C1FEFE00000002
polling failed 15

I am wondering how you got this because we have troubles to get any answers from the ds3. This is an answer anyway so my question is:
How you got this? Softwareversion? Are you able to reproduce it? What is your inverter's short id (inverter page status)
Thanks

iLeopod, 2022-03-12:
hi @patience4711 
as pairing always fail for me. I thought i just wasnt showing in the GUI and i tried polling withe the command `10;POLL=x: ` in console.
this i get when i repeat that steps now (10;INIT_N: + 10;POLL=1: )

```
zb send cmd 2
--
inMessage = FE064180020202020702C2 rc = 11
zb send cmd 3
inMessage = FE0166050062 rc = 6
zb send cmd 4
inMessage = FE0166050062 rc = 6
zb send cmd 5
inMessage = FE0166050062 rc = 6
zb send cmd 6
inMessage = FE0164000065 rc = 6
zb send cmd 7
inMessage = FE00660066FE0145C0088CFE0145C0088C rc = 17
sending NO cmd
inMessage = FE0145C0088CFE016401C2A6 rc = 12
inMessage = FE0E670000FFFF80971B01A3D8000007080010FE0145C0098D rc = 25
starting ZB coordinator failed
poll inverter 1
zb send poll cmd inverter 1
polling answer FE0164010064FE034480001401D2FE274481000006000000141400FF0000000000001380971B01A3D8FBFB06BB000000000000C1FEFE00000002
no route receipt
ignoring, received FE0164010064FE034480001401D2FE274481000006000000141400FF0000000000001380971B01A3D8FBFB06BB000000000000C1FEFE00000002
polling failed 15

```

iLeopod, 2022-03-12:
please note that im still not able to pair  different DS3-S with this firmware https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.0.x/bin im thinking of reflashing my coordinator with [kadzsol]s firmware

patience4711, 2022-03-14:
@iLeopod  
I cant make much out of this. When starting the coordinator fails, pairing / polling will also fail. 
If the coodinator fails to start there is something wrong with your zigbee module/wiring/firmware. With other firmware this won't work either.

iLeopod, 2022-03-14:
> @iLeopod I cant make much out of this. When starting the coordinator fails, pairing / polling will also fail. If the coodinator fails to start there is something wrong with your zigbee module/wiring/firmware. With other firmware this won't work either.
healthcheack and starting coordinator for pairing seams fine:
```
check zb system
inMessage = FE02610179071C rc = 7
ping ok
inMessage = FE0E670000FFFF80971B01A3D8000007090011 rc = 19
```

```
command = 10;INIT_P:
--
init zb coordinator
zigbee module reset
zb send cmd 0
inMessage = FE064180010202020702C1 rc = 11
zb send cmd 1
inMessage = FE064180020202020702C2 rc = 11
zb send cmd 2
inMessage = FE0166050062 rc = 6
zb send cmd 3
inMessage = FE0166050062 rc = 6
zb send cmd 4
inMessage = FE0166050062 rc = 6
zb send cmd 5
inMessage = FE0166050062 rc = 6
zb send cmd 6
inMessage = FE0164000065 rc = 6
zb send cmd 7
inMessage = FE00660066FE0145C0098D rc = 11
inMessage = FE0E670000FFFF80971B01A3D8000007090011 rc = 19
ZB coordinator started

```

patience4711, 2022-03-14:
@iLeopod  Yes, that is alright.   So the coordinator for pairing starts but the pairing itself always fails. And init_n , (should also end with FE0E670000FFFF80971B01A3D8000007090011) does that always fail?  
Btw, if you use 10;diag, you get more debug info.
We know we can pair a ds3, you have a DS3-S, i wonder if that would make a difference. We also know that we have to investigate how we can trigger a poll, so far no results.

lumbsta, 2022-03-15:
@patience4711 
Before starting sniffing, I tried the discord scripts with my newly purchased CC2531 and I think I got some kind of answer from the inverter:

> 08:27:29.22 COM5 AF_INCOMING_MSG (0x4481)
    GroupID: 0x0000
    ClusterID: 0x0106
    SrcAddr: 0x51C9
    SrcEndpoint: 0x14
    DstEndpoint: 0x14
    WasBroadcast: 0x00
    LinkQuality: 0x1C
    SecurityUse: 0x00
    Timestamp: 0x0006AE58
    TransSeqNumber: 0x00
    Len: 0x69
    Data: p@... ..\.. ...................-.).~......."........C................................................8... (0x70, 0x40, 0x00, 0x01, 0x08, 0x20, 0xFB, 0xFB, 0x5C, 0xBB, 0xBB, 0x20, 0x00, 0x03, 0x00, 0x09, 0xFF, 0xFF, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xA0, 0x00, 0x07, 0x0F, 0x06, 0xFB, 0x00, 0x2D, 0x00, 0x29, 0x03, 0x7E, 0x13, 0x88, 0x0B, 0xA2, 0x00, 0x1D, 0x00, 0x22, 0xFF, 0xFF, 0x04, 0xF8, 0x06, 0xAC, 0x00, 0x08, 0x43, 0xFF, 0x00, 0x07, 0x06, 0xCC, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x38, 0xB5, 0xFE, 0xFE)

patience4711, 2022-03-15:
@lumbsta  Indeed that is a good answer. It starts with your inverter serial.  I don't know what you exactly did, did you use the fake-ecu? We have seen this already i think, it's the same as what @jenswalit got, i think he used the fake-ecu.
Anyway, according to kadszol, the sniff should record the trafic of this successfull poll, and then the same for the (unsuccessfull) polling with the esp-ecu.

lumbsta, 2022-03-15:
@patience4711 
I used Z-Tool with the scripts provided inside the discord-11-7-2021 folder you posted [here](https://github.com/patience4711/read-APSystems-YC600-QS1/issues/19#issuecomment-1064028015).
Later I will prepare my 2nd CC2531 for sniffing, so I can hopefully supply the results by tomorrow noon.

patience4711, 2022-03-15:
@lumbsta Yes that is the so-called fake-ecu. So you have 2 cc2531 modules to work with. Great, then it should be doable. I hope that all your efforts will pay off.

patience4711, 2022-03-16:
@lumbsta good work! 
When i look at the polling requests it seems that they both have no answer, is that correct? I thought that what kadszol needs is a successfull poll trafic. I don't know what he expects to see in this information, for me it is secret language.  
To be sure, CC2531.psd is the snif of a pollrequest / answer of the fake-ecu (z-tool).
The other is the esp-ecu apparently with a cc2530.

lumbsta, 2022-03-16:
@patience4711 Thank you!
I guess the answer is in that pink block, because I saw it decrypted in the fake-ecu log at the same time. I'd probably have to use Wireshark to see what's behind.

Yes, CC2531.psd is the fake-ecu in Z-Tool and CC2530+ESP...obvious :D

At this point we might need some help from kadszol, as I am not familiar with the Zigbee protocol either.

patience4711, 2022-03-16:
@lumbsta So you are saying that the z-tool sniff was successful. Good, can kadszol produce those pictures with your zip? Than i can post it on https://github.com/Koenkk/zigbee2mqtt/issues/..

patience4711, 2022-03-17:
@lumbsta we are analizing your data, there are some differences but not yet an idea to solve the problem. To get things clear on what you did.
-You configured the fake-ecu, paired your inverter with it and could poll it. This trafic you sniffed
-Next you started the esp-ecu, did you pair it again? Or did you assume that this was not neccesary? 
This could be important because both coordinators seem to have different cluster id's.
Can i ask you to participate in the discussion here https://github.com/Koenkk/zigbee2mqtt/issues/4221 that is more efficient.

lumbsta, 2022-03-17:
@patience4711 

> -You configured the fake-ecu, paired your inverter with it and could poll it. This trafic you sniffed

Correct.
> -Next you started the esp-ecu, did you pair it again? Or did you assume that this was not neccesary?

Well, I tried to pair it again, but it failed. So I used 10;edit to provide the hex-address. An inverter reset might help. Will try in a few minutes.

> Can i ask you to participate in the discussion here https://github.com/Koenkk/zigbee2mqtt/issues/4221 that is more efficient.

Sure ;)

iLeopod, 2022-03-20:
fineally able to pair

```
received : FE0164020067FE1A448100000C0200001414013B0084550000000670300001929135EC0DAAFE0345C435EC005BFE1C44810000010135EC1414003B00BE5500000008FF1A70300001929135EC0EA8
--
pair command = 2524020FFFFFFFFFFFFFFFFF14FFFF140F0102000F1100703000019291A3D810FFFF80971B01A3D841
sending paircmd 4
received : FE0164020067FE25448100000F01000014140139009D6600000011703000019291A3D810FFFF80971B01A3D835EC0DB7FE0345C435EC005BFE1C44810000010135EC14140043001F6E00000008FFFF70300001929135EC0EAF
pair command = 1A24020FFFFFFFFFFFFFFFFF14FFFF14010103000F060080971B01A3D84F
sending paircmd 5
received : FE0164020067FE1A44810000010100001414013B0042790000000680971B01A3D835EC0D7A
success, inverter got id 0xEC35
sending NO cmd
inMessage = FE0164010064FE034480001401D2 rc = 14
inMessage = FE0E670000FFFF80971B01A3D8000007090011 rc = 19

```

iLeopod, 2022-03-21:
Cause i bought a cc2350 with CC2591 i think i had the wrong firmware and also used a onther flashing method. Other  point could be to less sun. Is there any way helping testing decoding the polling answers?

patience4711, 2022-03-21:
@lumbsta I understand that from your sniffs we can conclude that the esp-ecu sends the right query command, and the inverter responses. But how does this work out in the esp-ecu software? Do we see this response in the console when a poll command is sent? 
Any idea why it works now?

lumbsta, 2022-03-22:
@patience4711 
This is the console output during a valid inverter response:

> init zb coordinator
zigbee module reset
zb send cmd 0
inMessage = FE064180010202020702C1FE0166050062 rc = 17
zb send cmd 1
inMessage = FE064180020202020702C2 rc = 11
zb send cmd 2
inMessage = FE0166050062 rc = 6
zb send cmd 3
inMessage = FE0166050062 rc = 6
zb send cmd 4
inMessage = FE0166050062 rc = 6
zb send cmd 5
inMessage = FE0166050062 rc = 6
zb send cmd 6
inMessage = FE0164000065 rc = 6
zb send cmd 7
inMessage = FE00660066FE0145C0088CFE0145C0088C rc = 17
sending NO cmd
inMessage = FE0145C0088CFE016401C2A6 rc = 12
inMessage = FE0E670000FFFF80971B01A3D8000007080010FE0145C0098D rc = 25
starting ZB coordinator failed

And this is what I get while sniffing at the same time:
[image]
[image]

So maybe your software is misinterpreting the payload?

patience4711, 2022-03-22:
@lumbsta But this is about starting the coordinator. The answers we see have nothing to do with the inverter, it is serial communication from esp to zb module. You can't sniff that. 
So the question is why is the coordinator not starting. Are you sure about the software on the 25xx?

To be sure i just tested it with software version https://1drv.ms/u/s!AkMG5FvUwhediykfm8BZODfCHx5Y?e=kEeBgr
(ESP-ECU-DS3_TEST2) and the coordinator starts nicely at boot, like it should.

patience4711, 2022-03-22:
@iLeopod 
We are already made good progress with the decryption of the polling answers. Wat we would like is to compare the outcome with the figures of a APS ecu. If you have one, you could help i think.

iLeopod, 2022-03-23:
Unfortunatly i dont own an offical APS ECU.
I also got the starting ZB coordinator failed but i can still poll one of my inverters.

patience4711, 2022-03-23:
@iLeopod @lumbsta There was a small bug when using the console command 10;init_n . The response was sometimes "starting coordinator failed" but in fact it is running oke (10;health proves that). Sorry that this caused some confusion, this bug always existed but never appeared, no-one used this command ever.

When polling manual, you don't have to use this command. Polling fails when the coordinator is down.  When you just type 10;healt, the coordinator is checked and if not running it is started.
Normally the coordinator is started at boot. 
I removed the init commands in this version test3 https://1drv.ms/u/s!AkMG5FvUwhediysDkjdtNfUoPDfA?e=qoZGwW

lumbsta, 2022-03-25:
@patience4711 Finally found some time to test your software.

This is the console output after using the command: 10;poll=0

> poll inverter 0
zb send poll cmd inverter 0 cmd:1D2401C9511414060001000F1380971B01A3D8FBFB06BB000000000000C1FEFEB1
polling answer FE0164010064FE034480001401D2FE0345C4C951001AFE0345C4C951001A
no AF_INCOMING_MSG
polling failed 13

And this is the payload sent from the inverter after using 10;poll=0
[image]

So polling command is working 100%.

patience4711, 2022-03-25:
@lumbsta, This is good news. 
Unfortunately this is not the whole payload, the software cuts off everything before the 48 00.  The whole payload is displayed in the serial.hlml file. I got this serial.html from @jenswalit and the following is a complete poll answer. 
07:53:36.8 COM13 AF_DATA_REQUEST_SRSP (0x6401)
    Status: afStatus_SUCCESS (0x0)
07:53:36.8 COM13 AF_DATA_CONFIRM (0x4480)
    Status: afStatus_SUCCESS (0x0)
    Endpoint: 0x14
    TransID: 0x01
07:53:36.85 COM13 ZDO_SRC_RTG_IND (0x45C4)
    DstAddr: 0x6C22
    Relay Count: 0x00
    Relay List
07:53:36.85 COM13 ZDO_SRC_RTG_IND (0x45C4)
    DstAddr: 0x6C22
    Relay Count: 0x00
    Relay List
07:53:36.93 COM13 ZDO_SRC_RTG_IND (0x45C4)
    DstAddr: 0x6C22
    Relay Count: 0x00
    Relay List
07:53:36.94 COM13 AF_INCOMING_MSG (0x4481)
    GroupID: 0x0000
    ClusterID: 0x0106
    SrcAddr: 0x6C22
    SrcEndpoint: 0x14
    DstEndpoint: 0x14
    WasBroadcast: 0x00
    LinkQuality: 0x37
    SecurityUse: 0x00
    Timestamp: 0x00318A33
    TransSeqNumber: 0x00
    Len: 0x69
    Data: p0......\.. ...............M.[.....e....... ...5.#...................................................5/.. (0x70, 0x30, 0x00, 0x02, 0x13, 0x00, 0xFB, 0xFB, 0x5C, 0xBB, 0xBB, 0x20, 0x00, 0xFC, 0x00, 0x01, 0xFF, 0xFF, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x06, 0x4D, 0x06, 0x5B, 0x00, 0x10, 0x00, 0x12, 0x03, 0x65, 0x13, 0x89, 0x07, 0x89, 0x00, 0x05, 0x00, 0x20, 0xFF, 0xFF, 0x05, 0x35, 0x05, 0x23, 0x00, 0x01, 0xD0, 0x0A, 0x00, 0x02, 0x00, 0xA0, 0x00, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x35, 0x2F, 0xFE, 0xFE)
07:54:36.74 COM13 SYS_OSAL_TIMER_EXPIRED_IND (0x4181)

This is what appears at the serial port of the esp too.  
The whole response exists of 6 messages (the yc and qs have only 4), in the your polling answer from the esp we recognize only the first 4.
I see that there is a time gap between the 4th  (07:53:36.85) and the 5th 07:53:36.93  = 80 mSec
So maybe the serial reading stops here because it thinks nothing is available on the serial port anymore.

In this version i changed the serial reading function in such way that if nothing available it waits 120 mSec to see if something shows up.
 https://1drv.ms/u/s!AkMG5FvUwhediywW1gNDcpLC00vH?e=yBcg1s
Unfortunately i cant test these things but i hope it will work.

jenswalit, 2022-03-25:
@patience4711 
I am testing your new version with my DS3-S today

jenswalit, 2022-03-25:
@patience4711 
By polling i get this
`
<html>
<body>

poll inverter 0
--
zb send poll cmd inverter 0 cmd:1D2401226C1414060001000F1380971B01A3D8FBFB06BB000000000000C1FEFE67
polling answer FE0164010064FE034480001401D2FE0345C4226C00CCFE0345C4226C00CC
no AF_INCOMING_MSG
polling failed 13
zb send poll cmd inverter 0 cmd:1D2401226C1414060001000F1380971B01A3D8FBFB06BB000000000000C1FEFE67
polling answer
polling failed 50

</body>
</html>`

[107 more replies]

closed 2022-08-30 by patience4711