patience4711/read-APSystems-YC600-QS1-DS3 #20: Send all vales on MQTT

PLCHome, 2022-03-21:
It is possible to send all values ​​on MQTT.
Current, panel voltage, frequency, total power and mains voltage are only available on HTTP

patience4711, 2022-03-22:
@PLCHome  thank you for your interest in my software. The answer is no because it exceeds the max payload of mqtt messages.

PLCHome, 2022-03-24:
Do you think it would be possible to send a second and third MQTT message with the values? 
Either via new topics or via the same topic with other jason var names in teh message.

patience4711, 2022-03-28:
@PLCHome the program is stretched to almost the maximum size that can be used for ota updates, We like to keep some reserve for necessary extensions so to my regret no.

closed 2022-04-28 by patience4711

ehoutsma, 2022-05-23:
I changed AAA_DECODE and added the following lines which gives me all the output and sends it directly to mqtt.

`   MQTT_Client.publish (("ECU/Inverter"+String(which)+"/p0").c_str(), String(Inv_Data[which].power[0]).c_str());
   MQTT_Client.publish (("ECU/Inverter"+String(which)+"/p1").c_str(), String(Inv_Data[which].power[1]).c_str());
   MQTT_Client.publish (("ECU/Inverter"+String(which)+"/p_total").c_str(), String(Inv_Data[which].power[4]).c_str());
   MQTT_Client.publish (("ECU/Inverter"+String(which)+"/energy").c_str(), String(Inv_Data[which].en_total).c_str());
   MQTT_Client.publish (("ECU/Inverter"+String(which)+"/VDC0").c_str(), String(Inv_Data[which].dcv[0]).c_str());
   MQTT_Client.publish (("ECU/Inverter"+String(which)+"/VDC1").c_str(), String(Inv_Data[which].dcv[1]).c_str());
   MQTT_Client.publish (("ECU/Inverter"+String(which)+"/IDC0").c_str(), String(Inv_Data[which].dcc[0]).c_str());
   MQTT_Client.publish (("ECU/Inverter"+String(which)+"/IDC1").c_str(), String(Inv_Data[which].dcc[1]).c_str());
   MQTT_Client.publish (("ECU/Inverter"+String(which)+"/VAC").c_str(), String(Inv_Data[which].acv[0]).c_str());
   MQTT_Client.publish (("ECU/Inverter"+String(which)+"/Temp").c_str(), String(Inv_Data[which].heath).c_str());
   MQTT_Client.publish (("ECU/Inverter"+String(which)+"/Freq").c_str(), String(Inv_Data[which].freq).c_str());`

Also had to fix the total power counter, since that was empty:

`    power = en_incr_total / ts * (float)3600; //[W]
    dtostrf( power, 0, 1, Inv_Data[which].power[4]); // write the value in the struct
`

To save space there is a lot of HTML that can be shrinked/minified and save a lot of space doing that.

ehoutsma, 2022-05-27:
Arround line 303
`    power = en_incr_total / ts * (float)3600; //[W]
    dtostrf( power, 0, 1, Inv_Data[which].power[4]); // write the value in the struct

    Inv_Data[which].en_total += en_incr_total; // stack the increase`

I also store the energycounter per panel. Nice to see the difference between the panels.

kpwg, 2022-08-06:
@ehoutsma 
Would you like to share the code includes shrinking html for this or the .bin? I'm currently playing with openDTU and am amazed at how much data comes via MQTT :)

github-iw, 2022-08-17:
@ehoutsma, this would indeed be a great improvement on this already very nice software package from @patience4711 
Maybe also possible to add autodiscovery to the mqqt messages at startup?