patience4711/read-APSystems-YC600-QS1-DS3 #21: CONSOLE Mixed content exception behind reverse proxy on HTTPS

PLCHome, 2022-03-21:
After the reverse proxy the webbrowser will rais an exception:

```
CONSOLE:72 Mixed Content: The page at 'https://xxxx.de/CONSOLE' was loaded over HTTPS, 
but attempted to connect to the insecure WebSocket endpoint 'ws://xxxs.de/ws'. 
This request has been blocked; this endpoint must be available over WSS.
```

can you change the follow line in the CONSOLE file?

```
var gateway = `ws://${window.location.hostname}/ws`;
```

to 

```
var gateway = `${(window.location.protocol == "https:"?"wss":"ws")}://${window.location.hostname}/ws`;
```

so it would call wss on https and ws on http

patience4711, 2022-03-22:
@PLCHome thank you for your interest in my software.
I changed the line you suggested and tested it. 
In the next version, (adapted for the ds3 inverter) it will be included.

closed 2022-03-28 by patience4711

PLCHome, 2022-04-01:
@patience4711  Tested: OK
Thanx