patience4711/read-APSystems-YC600-QS1-DS3 #23: No data after power down and power up

PLCHome, 2022-04-01:
Yesterday around 11:00 a.m. I switched the device and did not check whether data was being retrieved.
After I clicked on pair inverter today, data is coming in again.

How is that thought? Why can't the ESP reconnect to the inverter after startup?

```
system time = 14:46 hr.   summertime
firmware version : ESP-ECU-v8_0
time retrieved today : yes
the signalstrength of the connection to the router = -33
de mqtt clientId = : 2xxxx
status mqtt : connected to 1xxx
system up time: 0 days 5 hrs 57 min.
current errorCode = 0
polling starts at 7:24 and stops at 19:53
PAIRED INVERTERS
inverter 0 serialnr 8xxxx ID = 0xCxxx

ESP INFORMATION
ESP CHIP ID nr:	2xxxx	Flash Chip ID nr:	1xxxxx
IDE Flash size:	4194304 bytes	Real Flash size	4194304 bytes
Free memory	17344 bytes	remote IP	xxx
variables dump
value=0 inverterCount=1 zigbeeUp=1
pollRes= 300 now=1648824375 switchonTime=1648797880 switchoffTime=1648842810
unixtime=1648824375
polled = 100000000
resetCounter = 0
Content filesystem :
Inv_Prop0.str size 48
basisconfig.json size 91
mqttconfig.json size 168
wificonfig.json size 84
```

PLCHome, 2022-04-01:
I just found the ESP_ECU-9.0 update. 
I will test the scenario again.

patience4711, 2022-04-01:
@PLCHome  What ESP-ECU version? You have DS3 inverter? 
I don't know what happened. Normally you don't have to pair the inverter again. If you power-switched, then the zigbee network goes down. The zb module needs to restart too, but after that everything should work. 
If the software is updated, the esp reboots too and no reports of this malfunction have reached me. 

Maybe coincidence.

PLCHome, 2022-04-02:
It is an QS1
After power down and up:
[image]
After reboot:
[image]
After repair:
[image]

i just installed 9.0 bevor testing again.
```
system time = 12:58 hr.   summertime
firmware version : ESP-ECU_v9_0
time retrieved today : yes
the signalstrength of the connection to the router = -27
de mqtt clientId = : 229795
status mqtt : connected to XXX
system up time: 0 days 0 hrs 12 min.
current errorCode = 0
polling starts at 7:22 and stops at 19:55
PAIRED INVERTERS
inverter 0 serialnr 8XX ID = 0xXXXX

ESP INFORMATION
ESP CHIP ID nr:	XXX	Flash Chip ID nr:	XXX
IDE Flash size:	4194304 bytes	Real Flash size	4194304 bytes
Free memory	17104 bytes	remote IP	XXX
variables dump
value=0 inverterCount=1 zigbeeUp=1
pollRes= 300 now=1648904304 switchonTime=1648884147 switchoffTime=1648929307
unixtime=1648904304
polled = 100000000
resetCounter = 0
Content filesystem :
Inv_Prop0.str size 48
basisconfig.json size 91
mqttconfig.json size 168
wificonfig.json size 84
```