patience4711/read-APSystems-YC600-QS1-DS3 #26: Problem with my CC2531 USB stick does anyone have an idea

redseasix, 2022-04-21:
Hello all,

I am not sure if the CC2531 works properly at all.
I have written the firmware: **CC2531ZNP-with-SBL.hex** from the ZIP file to the USB stick with the Flash Tool from TI. The verification within the TI tool said OK.
When poking the LED lights up for a while then it goes off on the stick and comes on after a while.
Now I am in the console and have there: 10;ZBT=2710AABBCC entered and as a response this received: 
[image]
Does that mean that the stick was recognized correctly? Or should there be something after answer?
Do I understand correctly "starting ZB coordinator failed" that means that it is not properly there.
I use this CC2531-USB stick:
Does that mean that the stick was recognized correctly? Or should there be something after answer?
Do I understand correctly "starting ZB coordinator failed" that means that it is not properly there.
I use this CC2531-USB stick:

[image]

Here is my wiring

ESP         ZIGBEE USB STICK
GND    ->   PIN 1 GND
3,3 V    ->   PIN 2 VDD
D8        ->  PIN 3 Debug Clock
D7         -> PIN 4 Debug Data
D5         -> PIN 7 Reset

flat cable are 1:1

[image]

[image]

I also flashed a 2nd USB stick and got the same result.

could you please give me a hint, where I can look further, what I am doing wrong?

Thank you redseasix

patience4711, 2022-04-21:
@redseasix  you are not getting reponse so either the device or the serial communication doesn't work. You could try to exchange rx and tx

SteffenB71, 2022-04-22:
Hallo
cc2531 -> esp (nicht von mir getestet!)
P0.1 -> d8
P0.3 -> d7
Rst-> d5
vcc -> 3,3 V - 2
gnd -> 1
The CC2531 is treated like the CC2530. The firmware that is flashed can also be from CC2530.
Steffen

redseasix, 2022-04-26:
Hello Owner patience4711

Thanks for your feedback. your hint put me on the right track and the Zgbee stitch now answers correctly to 10;ZBT=2710AABBCC.

What did I do:

Programmed Zigbee firmware CC2530ZNP-with-SBL.hex from the folder CC25xxfirmware ds3 to my Zigbeestick CC2531.
now it responds correctly

cc2531 -> esp
P0.2     -> d8
P0.3     -> d7
Rst       -> d5
vcc       -> 3,3 V
gnd      -> GND

IO Connector of Zigbee Stick:
[image]

[image]

here is my working wiring

[image]

now it's time for the next steps

redseasix, 2022-04-26:
super that was the starting signal for me. The Zigbee stick responds exactly as described.

now I have to look that I get the ESP with the CC2531 stick close to my two QS1. (Gatenhaus about 80m away (Bad WLAN with -81dBm So I have to build me another WLAN repeater.

in 3 weeks comes another DS3 which is already here with 2 solar cells a 200 watts to the start.

please adapt in the README.md still the allocation for the CC2531 Zigbee stick:

**cc2531 -> esp**
P0.2 -> d8
P0.3 -> d7
RST - > d5
VCC -> 3.3 V
GND -> GND

Thanks for the help

closed 2022-04-28 by patience4711