patience4711/read-APSystems-YC600-QS1-DS3 #27: Problem with compiling code with Arduino IDE

homeautomation2022, 2022-04-23:
I would like to know what version of Arduino IDE the source can be compiled with?
I use the latest stable version (1.8.19) and have also tried the esp8266 board library version 2.7.1 as described in the readme, as well as the latest one.
I installed the required libraries afterwards, so far no problem, but the Arduino IDE complains quite early on with the code:
#include <sunMoon.h> needs to be changed to #include "sunMoon.h" because it's not a library but a separate header file in the project and all included HTML header files need to have a lower case h instead of a upper case, otherwise comes the file was not found.
I would like to reduce the code to a minimum (with the stable base that I get the data from) and then do something with it myself, so that I can evaluate the data myself and at the same time be able to do something else with this ESP, but here it is I have to get the program up and running first. The binary is nice for everyone else who wants it easy, but unfortunately doesn't allow for their own code adjustments.

I just want to reliably read a YC600 and have flashed the new cc2530 firmware, that should work, maybe a little better???
If someone already has a fairly simple code for this, I would be grateful for that as well. I've also found the yc600_serial library in PlatformIO, but I think this one is better developed in terms of reliability.

patience4711, 2022-04-23:
I compile with arduino 1.8.13

homeautomation2022, 2022-04-24:
Thank you for your short information, I tested this version immediately, of course, with the result that the same errors come.
The include problem is quickly fixed, but then the linker throws up a few more warnings and errors in ZIGBEE_HELPERS.ino, TIJD_CALC.ino and sunMoon(undefined reference)?.
I have no idea what the simple trick is that it just works for you.
I can attach the error output here, but it's probably easier to test it on a PC on which no Arduino IDE was installed, since the settings and libraries are saved across versions.

I had to install the following libraries:
ArduinoJson
ESPAsyncTCP
ESPAsyncWebServer
NTPCIient
PubSubClient
Time
Maybe there is a difference, but it currently points more to the code than a library.

patience4711, 2022-04-24:
Sorry, i can't investigate what the problem is, and to be honest i am not very interested to spend time on that . I can compile this code so i am sure you'l find a way to compile too. 
Good luck

homeautomation2022, 2022-04-26:
I commented out the SunMoon.h and in the TIJD_CALC.ino also commented out the calls and set the variables sunset and sunrise manually for testing, then I can at least build and flash, according to the include adjustments already written.
However, I once loaded this project into PlatformIO and found that there were a few more errors, e.g. bool functions without return. The code doesn't seem to be very clean, but first of all I was only concerned with having a working basis.
Unfortunately, I then had to realize that neither the finished bin nor my adapted version found the inverter. The seller of the module has already flashed the desired new variant for the CC2530, but since I couldn't check that, I wrote the old firmware from 2021 for the CC2530 on it again, since I only want to connect the YC600 anyway, but that unfortunately nothing has changed. That annoyed me at the end of the weekend, so I stopped there.
I think the other project YC600_serial from runnigpenguin is programmed more simply and cleanly, even if the code is a bit outdated, I'll probably continue there, I was able to load it, but I couldn't find the inverter either. It's either the firmware, the reception, or the inverter transmitter might be dead???
I thought it would be easier with already finished projects :-/

closed 2022-04-26 by patience4711

homeautomation2022, 2022-05-07:
Does anyone here have compilable source code?
Unfortunately, I get from one error to the next and have no idea how @patience4711 can compile this code, it is full of warnings and cannot be compiled at all without modifications.
After fixing the errors, I was able to flash, but despite entering the correct WiFi data and a successful connection, the ESP restarts in the portal.
Unfortunately, the code is very opaque for me with Dutch comments and very tinkered with the indentations and commented out lines.

maps2002, 2022-05-08:
Hi @homeautomation2022 ,

On Windows, with Arduino 1.8.19 I am able to compile with minor adjustments:
 - Rename (in the filesystem) header files (like PORTAL_HTML) from upper ".H" to lower ".h". Example: PORTAL_HTML.H to PORTAL_HTML.h
 - Make sure you manually install the sunMoon library in Arduino: https://github.com/sfrwmaker/sunMoon

After that I was able to compile it.

As to the Wifi issues, I am able to have better success connecting if I increase the delay/retries to connect in START_WIFI.ino, function start_wifi. For example increase to 30 events and/or increase the delay between retries.