patience4711/read-APSystems-YC600-QS1-DS3 #29: INFOPAGE panel 1&2 values swapped

karlheinz2000, 2022-05-13:
On INFOPAGE values from physical input 1 are reported as dc volt2 and dc curr2 and vice versa.

On "panel overview" page and over mqtt values are reported correctly.
firmware version : ESP-ECU_v9_3

For testing only one input was used, other input was open.
Sceenshot is with input 1 used.

[image]

patience4711, 2022-05-13:
@karlheinz2000 It would be really helpfull to mention what type inverter you have.
The physical input 1 you mention, is that as such indicated on the inverter? How is the other input indicated, as 2 or 0? Just to be shure what is swapped.

karlheinz2000, 2022-05-13:
It is DS3-L with inputs labeled as 1 & 2.
Connected was input 1 during this test.

[image]

patience4711, 2022-05-13:
Oke, clear. In this version i swapped these values.

https://1drv.ms/u/s!AkMG5FvUwhediz9mkWZeSLpUUuAo?e=iAn0eb

Please report back if this works. 
Another question: If you powercycle your ESP/ZB module, does the polling still work?

karlheinz2000, 2022-05-13:
**Test passed!** Values are now consistent in all outputs.

I did not power cycle much in the past. ESP was running almost 2 weeks with no restart.
I'll set up a power cycle test begin of next week.

Currently I try to verify the "correction factor" for my DS3-L in a static power supply setup. Also the DC current value needs correction...

homeautomation2022, 2022-05-13:
If I remember correctly, the values on my YC600 are also swapped (one panel is very often in the shade...), but I can check again tomorrow.

patience4711, 2022-05-16:
@karlheinz2000  about the channels there was a lot of discussion in the past, It is considered to be correct, But if not you can always swap the wires.

karlheinz2000, 2022-05-16:
Ok, I don't know about discussion in the past. 
For DS3-L **power** values are reported **correct** in mqtt and web. All is ascending. Input "1" reports to mqtt "p0" and web "panel 1", input "2" reports to mqtt "p1" and web "panel 2".
For DC **voltage** and **current** this was **not consitent** as in the [screen shot](https://github.com/patience4711/read-APSystems-YC600-QS1-DS3/issues/29#issue-1234892677). Input "1" was reported as "volt2/curr2" and input "2" was reported as "volt1/curr1". 
It is not really important for me, you can go back, if needed. Only power matters and this is correct.

patience4711, 2022-05-16:
@karlheinz2000  Yes, but i would say only energy kWmatters as this is what the inverter reports. (And where we get our pay-back from). The power (kWh) is calculated from the energy value.

karlheinz2000, 2022-05-17:
@patience4711 You are right, only energy (kWh) matters. But it is nice to see peak power :-)

homeautomation2022, 2022-05-17:
Can anyone else confirm that the YC600 has panels 1 and 2 swapped?

patience4711, 2022-05-17:
@homeautomation2022 if you turn the inverter upside down the 1 is on the left, i don't know the logic behind it. So it is arbitrary which is 1 but as long as you know which is which, it can't be a problem i think. 
I close this issue now, as i said, the ones who reversed engineered the yc600 made the conclusion that this is it.

closed 2022-05-17 by patience4711