patience4711/read-APSystems-YC600-QS1-DS3 #30: Can't add inverter

StiffelMS, 2022-05-13:
Hi everyone!

I just received my nodeMCU and CC2530, Flashed "ESP-ECU_v9_3" and "CC2530ZNP-with-SBL.hex" for reading out my APSystems QS1.

In webinterface, I can't save/pair any inverter. Is the "pair-button" the "save-button" in the button right corner?
Every time I fill out the inverter-add-page and click save, the nodeMCU reboots...

Any ideas? Thanks!

StiffelMS, 2022-05-15:
I flashed nodeMCU again, now it saves data. 

What is the username and password (the pw must be 1111 (standard) right?)? I can't log in...

pixeldoc2000, 2022-05-16:
@StiffelMS 
I believe it is user: "admin" and password: "1234" by default ?

patience4711, 2022-05-16:
@StiffelMS  It is 0000 you should have seen that in the wifi configuration access point.

patience4711, 2022-05-16:
@StiffelMS its 'admin'and 0000 or 'user' and 1111 but the user cannot change settings.

closed 2022-05-16 by patience4711