patience4711/read-APSystems-YC600-QS1-DS3 #31: Callibration factor for DS3-L

karlheinz2000, 2022-05-16:
This is to report the calibration factor for **DS3-L.**
I did bench measurements with DS3-L connected to a power supply in cc mode. AC power was recorded with a LMG450 power meter.
DC voltage was set to 30V, DC current was adjusted to get the desired AC power in 50W steps from 50W to 350W.
Only one input was connected.

[image]

Out of this table **1.66** is a good match. Below 150W reported power is not linear.
I verified this factor with 2 DS3-L. 

Also DC current needs callibration. The reported values on INFOPAGE are not correct. DC current was measured as part of power measurement from above (50..350W).

[image]

A factor of **1.25** gives good results.

I wonder if these factors are also transmitted via zb? How does a APSystems ECU calculate correct values?

patience4711, 2022-05-16:
@karlheinz2000  Thanks for taking the trouble to get this factor right.
What is transmitted by ZB you can see in the polling answer. That gives us a value we have to decript into a correct energy value, that is what the software does. 
Did you conduct these tests to both channels?

karlheinz2000, 2022-05-17:
Complete set was done for one channel only, but I verified one point on 2nd channel.

lumbsta, 2022-05-17:
@patience4711 I can confirm @karlheinz2000 's power conversion factor of 1.66 with 2 modules attached. My daily solar production is now almost equal between ESP ECU and Shelly 1 PM.

patience4711, 2022-05-17:
@lumbsta  Oke, now we are getting somewhere..

closed 2022-06-20 by patience4711