patience4711/read-APSystems-YC600-QS1-DS3 #34: Can't add a second inverter

stephanpanholz, 2022-05-19:
When I want to add a second inverter, it doesn't add a new one but always overwrites inverter0.

Thanks in advance...

karlheinz2000, 2022-05-19:
@stephanpanholz 
Did you click "save" on bottom right?
I can add 2nd inv. 
Tested right now.

stephanpanholz, 2022-05-19:
Yep sure…i clicked Add, then entered the values for the new inverter and then I clicked save. But instead of creating a new one, my settings of inverter0 are overwritten. Which version do you use, 9.4?

stephanpanholz, 2022-05-19:
What I noticed is that, after clicking add, behind the word „inverter“ in the upper legt there is {Nr} indicated, instead of a new number e.g. 1. maybe this is the problem.

karlheinz2000, 2022-05-19:
I'm still using the version from here:
[https://github.com/patience4711/read-APSystems-YC600-QS1-DS3/issues/28#issuecomment-1128789105](https://github.com/patience4711/read-APSystems-YC600-QS1-DS3/issues/28#issuecomment-1128789105)

I'm also have {nr} and it works.

stephanpanholz, 2022-05-20:
Mmh when I follow the link it leads me to version 9.4.
I also tried 9.3 Beta3 with the same issue. What I will do now is, trying an older version if there the problem also persists.

stephanpanholz, 2022-05-20:
Now I tried version 9.2 and 9.0 with the same results. 
slowly I am doubting myself, maybe I am simply to stupid to check that?
Here again the steps I do:
1. Click on "Menu" and "Inverters"
2. Now I see the settings of inverter0
3. Click on "Add" on the top left
4. Enter the settings in the fields with a hight Dom. idx
5. Click on Save at the bottom right
6. A processing screen appears
7. i am redirected to the inverter overview where still inverter0 is displayed, but with the settings I've entered for the new inverter

karlheinz2000, 2022-05-20:
I do the same. But in 7. I get redirected to the main 4-panel-power-overview page. Not the INV page. 
I'm only testing. I have no idea about the SW behind :-(

stephanpanholz, 2022-05-20:
sorry I have to correct myself...I also get redirected to the overview page. but then the inverter0 has the new name etc.
okay I see...lets wait what patience4711 has to say...

patience4712, 2022-05-20:
@stephanpanholz  
What ESP device do you use? Is it new?
In step 7 there should be a menu item inv2.
I think maybe your filesystem is corrupted. Are all other settings correctly saved? 
Try to delete the existing inverter, if you check the infopage / vardump the invertercount should be 0. 
There should be no inverter configfiles, see filesystem down in the infopage. 

When nothing helps, try to erase the flash before re-flashing the ESP-ECU.bin on the ESP.

stephanpanholz, 2022-05-22:
So...I erased the esp8266EX completly and flashed it again. now it works to add a second inverter correctly. Also pairing works most of the time for both inverters but now I am having a new issue. only the values of one inverter are displayed, exactly the values of the inverter a paired at last. they alwas run into inverter0 and the second one displays n/a.
do you have an idea what the problem could be?

thanks in advance :-)

stephanpanholz, 2022-05-22:
when I try to poll the second inverter I get an "no AF_DATA_CONFIRM" message. I am using DS3 inverters, I forgot to mention, sorry.

ehoutsma, 2022-05-24:
I also had issues pairing the inverter. I restarted the inverter (covered the solarpanels and disconnected the AC line). After that there was no issue pairing the inverter. DS3-L btw.

closed 2022-05-27 by patience4711