patience4711/read-APSystems-YC600-QS1-DS3 #35: Adjustable poll rate

ehoutsma, 2022-05-23:
Is it possible to make the poll rate adjustable. I only have one DS3-L and would like to see a more frequent polling of the inverter. I've changed the pollrate to 60 seconds which works perfectly. Only one or two reinitializations yesterday.

Making this adjustable creates real added value over an ECU.

It also shows that an micro inverter isn't a perfect solution for a lot of systems, but that's a different story.

patience4711, 2022-05-26:
@ehoutsma Thank you for your interest in this software. In the past we had adjustable polling rate, i removed that because it doesn't add anything of value and causes the zigbee module to crash. If you want to poll it more often you can disable the automatic polling and use the api or mqtt to trigger a poll.
But you already solved it yourself.

closed 2022-05-26 by patience4711

ehoutsma, 2022-05-30:
@patience4711 Thank you for the great effort of building this software. It's a great option to customize and integrate these things into my home automation system. Especially if it would be possible to figure out how to limit the inverters, so they can limit the amount of power that is being returned to the net. https://emea.apsystems.com/wp-content/uploads/2017/06/APsystems_EMEA_WhitePaper_Export-limiting-functions-with_Microinverters_2017_P.pdf

If you have 10 inverters you want to read, sure, but if you only want to read 1 or 2 inverters, there is no need to wait 5 minutes for a update. Especially if you want to use the most of the power yourself or want to diagnose things.

The crashes of the zigbee module is a different issue. But a health check every so often is a fine sollution. Makes sure that the zigbee network is running nicely.

I'm running a poll every 10 seconds on a single DS3-L now, and it works really well, every now and than it stops, but the healthcheck fixes that in 2 minutes. (Was running at 60 seconds and the healthcheck then runs at half the pollrate). Switching to an external source for the pollingrate seems overly complicated. Making this system dependant on other machines to poll the data is not the prefered sollution, Especially since the pollRes is adjustable by uncommenting a few lines.