patience4711/read-APSystems-YC600-QS1-DS3 #36: Inverter temperature

Aresitoo, 2022-05-27:
Got it all working with 4 DS3 inverters, total outputs still are off from Shelly EM and a Tuya meter I got between the lines. Shelly EM total today 21.1KW and total sum from the ECU gives 22.65.
Also the temperature readings are certainly not correct, not more than 40 degrees while in the full Madrid sun and only 1.5cm under my panels, a similar instalation of a friend with the APsystems ECY reads alwasys about 15/20 degrees more and being further away from the solar panels

patience4711, 2022-05-28:
@Aresitoo We are fully aware of that, some values still have te be verified. If you read the issues about 'conversion factor' and 'testing ds3', you see that investigations are done on this. 
Maybe you can contribute by measuring the ambient temperature at sunrise and compare that to the value the inverter is reporting.

karlheinz2000, 2022-06-01:
@patience4711 , @Aresitoo 
I verified the temperature measurement. 
Four external sensors (DS18B20) attached to the inverter front and back side and ambient. All signals are logged during day. The sensor on back side of inverter reports almost same temperature as DS3. This is also consitent in terms of DS3 construction.
**For my DS3-L, the reported temperature from ECU is correct.** Refer to blue and red graphs.

Unfortunately outside temperature did not go above 25°C, so I can not verify 'full Madrid sun' ;-). I will keep logging running until batteries are empty. Hopefully summer temperatures will come..

[image]
.

patience4711, 2022-06-02:
@karlheinz2000 Thanks, you really made work of it. Surprisingly it wasn't so bad after all. 
I see at most some 15 degrees over ambient temperature at the hottest part of the day. I think that we can expect this to be more or less persistant. So 55 degrees in madrid could be realistic, that is nothing to worry about.

closed 2022-06-20 by patience4711