patience4711/read-APSystems-YC600-QS1-DS3 #39: modbus register of a DS3 / gridprofile settings helpfull?

hschraep, 2022-06-27:
Hello found this settings within a german forum. I think this could be helpfull for this project. Not sure if this register are exists also over zegbee but its a try. With the modbus register it is possible to set the output power of the DS3. The other thing are the gridsettings
for the countrys. Within the first pdf are the settings for germany. 
you will find the files here:
https://www.photovoltaikforum.com/thread/173721-apsystems-wechselrichter-mit-modbus-auslesen-und-beschreiben-yc600-qs1-ds3-usw/

lumbsta, 2022-07-05:
I would love to see this feature in a future release! Since the ECU has to pass the Modbus registers to the inverter via Zigbee, some sniffing would certainly be necessary. Unfortunately, I don't have an ECU.
@patience4711 Would you be interested in working on this?

hschraep, 2022-07-07:
@patience4711  What is needed to sniff the ECU traffic? I have a ECU-B now, but i have no raspberry it cost to mutch at this time to by it now. This is why i was interessted in this project but if the space is empty then he is empty :-)

patience4711, 2022-07-12:
For sniffing you need a cc2531 zigbeemodule. But as i remember well, you can't issue the relative commands with an ECU-B so you have to find someone wo owns an ECU-R and is able to sniff. If you go through  [this thread](https://github.com/Koenkk/zigbee2mqtt/issues/4221) you will find more info and persons who are maybe willing to help.
$lumbsta is able to tell you more on this subject. I have no experience with this sniffing.

gurkc006, 2022-09-01:
Hi together,
I own a ECU-C and six DS3 inverters and they are running fine. For the future I would love to use a raspberry/ESP+CC2530 to readout and control my inverters. At the moment I use the (hidden) webinterface to readout live data and to import it into my FHEM home automatisation. That works quite fine. The week after next I would like to build up my ESP+CC2530 setup and check, if I can sniff some of the gridcode messages and/or the standby and power throttle commands accessible via modbus TCP/RTU. There are some problems, as I managed to switch on/off the inverters using Modbus, but only all together. I couldn't switch of the inverters individually... So reporting back, when I have some successfull sniffs :-)

gurkc006, 2022-10-02:
Hi it's me again!
I successsfully setup my sniffing enviroment (cc2531 with TI firmware and TI PACKET SNIFFER, ECU-R, 6x DS3, rpi-ecu with H-2 Z-002 (TICC2530 CC2591) and firmware 'CC2530ZNP_2591-with-SBL.hex'). I can see the zigbee messages between original ECU-R and DS3 inverters. I also sniff the messages for power throtteling and switching inverters on/off. I also got the rpi-ecu running (when ECU-R is switched off).
Now I would like to generate and send messages to test the power throttle messages, but I have problems receiving the messages send by the rpi-ecu at all. The rpi-ecu is 15 m away from my living room (the inverters and pv modules and rpi-ecu is outside ob the roof of our garden house). 
So I'm a little bit lost here. I have seen, that the rpi-ecu seems to be not updated so regularly like the ESP-ECU?! 
@patience4711: would it help, if I send you the sniffs? I already discussed them with kadzol on discord, maybe you would like to join there?

patience4711, 2022-10-05:
Hi @gurkc006  Good work! If i understand wel, you sniffed the switching on-off messages and answers from the inverters. So now you are and you are trying to send such message via the rpi-ecu console. This should not be a problem. After sending, the RPI-ECU listens for an answer which should look like the ones you sniffed. 
You have problems receiving the rpi-ecu messages, with wat? With your sniffer?
You say the rpi-ecu is working, meaning that the pairing and polling is working oke? 
The RPI_ECU is completely up-to-date, that is not the issue here. In the journal you can see the signal quality, if that is poor you can try to move te rpi-ecu closer to the inverter.

I don't own a DS3 so sending me the sniffs is pointless. I am sure that @kadsol pointed you to the right form the message should have. If i know the commands i am happy to extend the program with these maintenance commands.

seulbdoowle, 2022-12-04:
> Hi it's me again! I successsfully setup my sniffing enviroment (cc2531 with TI firmware and TI PACKET SNIFFER, ECU-R, 6x DS3, rpi-ecu with H-2 Z-002 (TICC2530 CC2591) and firmware [...]

seulbdoowle, 2022-12-04:
Hi @gurkc006 nice work! I own a DS3 and would be interested in the sniff data for power throttle. The background is to try to set the power without an ECU unit. I would therefore be grateful to receive this information. However, I am currently waiting for the CC2530 so that I can communicate with the inverter. Maybe I will use a CC2531 in the meantime but I am afraid of the soldering work. Many thanks in advance

lukeIam, 2023-02-02:
what's the current status here? Is there a sniff containing the read/set of a modbus register?
I don't own a ECU but I would try to help analyzing the dumps.

closed 2023-05-25 by patience4711

lukeIam, 2023-05-29:
@patience4711 is this really completed? could not find it in the changelog...

Jeroen88, 2023-07-06:
> Hi it's me again! I successsfully setup my sniffing enviroment (cc2531 with TI firmware and TI PACKET SNIFFER, ECU-R, 6x DS3, rpi-ecu with H-2 Z-002 (TICC2530 CC2591) and firmware [...]

Hi @gurkc006 could you please share the messages you sniffed? If possible please tell for each message the direction (from rpi to inverter / from inverter to rpi) and the intention (throttle, switch on, switch off etc).If you also know the significance of each byte in the message that would be very nice!

gurkc006, 2023-07-09:
Hi [Jeroen88](https://github.com/Jeroen88)!
Sorry, but I tried to unterstand my "recordings" from 2022 but couldn't sort it out. The whole theme was very new to me and I didn't saved the sniffs in an ordered manner to disk. So I'm afraid I coudn't provide any sniffs from 2022. At the moment I use a small python script to control the DS3 output power (of the system at once) using Modbus TCP messages to the ECU and writing to the registers known. But I havent't any sort of access to the zigbee level messages itself at the moment. No sniffing hardware available. Sorry.

Jeroen88, 2023-07-09:
@gurkc006 thanks for your answer, too bad you did not keep the sniffings! Could you share the python script? It might be useful for someone else that has sniffing hardware and an ECU!

gurkc006, 2023-07-10:
Sure, for anyone, who would like to control DS3 output power via ECU. You have to enable Modbus in ECU settings and assign Modbus addresses to your DS3 units. In the script, enter ONE of your Modbus addresses to set total power of your micro inverters. Each execution of the script takes around 4-6 seconds. Setting of power is very quick then. Remember, after switching on, it takes 1-3 minutes to enable power generation due to syncing to power grid.
Have fun with it: [APSystem_setInverterPower.zip](https://github.com/patience4711/read-APSystems-YC600-QS1-DS3/files/11998435/APSystem_setInverterPower.zip)

Jeroen88, 2023-07-10:
@gurkc006 Thank you for sharing!

Pmachat, 2023-07-28:
> Sure, for anyone, who would like to control DS3 output power via ECU. You have to enable Modbus in ECU settings and assign Modbus addresses to your DS3 units. In the script, enter ONE of your Modbus [...]

Hello, 

does this script work with all ECUs (B, C, R)?
Where can I find more information?

Best regards

gurkc006, 2023-07-28:
Sorry, don't know. I have tested it with my ECU-R. 
The 5875 is a factor for my pv setup. I have 6 DS3-EU with nominal output power of 6x880VA = 5280Wpeak. The power setting is in promille. So 1000 promille = 5280W. But by try an error I found, that a factor of 5875W/1000promille fits better to my setup. You might adjust the factor for your setup. Hope that helps?

Pmachat, 2023-07-28:
thanks for these informations
I thought this WMaxLimPct value was the max value of a single solar panel and not of all the panels so in your example it was 8800 ( if your panel power is 880Wc)
Can you tell me what the power (Wp) of one of your solar panels?
Thanks

gurkc006, 2023-07-28:
I have six DS3-EU each with 2 Strings of 4 PV-Panels of 130Wpeak parallel = 48 panels á 130Wp = 6240Wp panel power.

Pmachat, 2023-07-28:
ok each stings= 4*130Wp=520Wp -> 2 string by DS3= 1040Wp -> 6 DS3= 6*1040=6240Wp.
what power value did you configure with the script for your installation?

gurkc006, 2023-07-28:
??? Sorry, I didn't understand your question? I wrote the script and found, that the gain factor of 5875 fitted quite well to my setup. So you will have to adapt this gain to your setup. On a sunny day, try setting the limit to e.g. 100 promille = 10% and get the output power of your DS3s. Then you know, 10 times this power will be 100% = 1000 promille. Set this factor in the script and you should be fine. Does this help?

Pmachat, 2023-07-29:
Thanks for this help
To run the script, I thought you had to type the command
python APSystem_setInverterPower.py --power xxx with xxxx the desired value
The script then calculates iPower = max(min(int((args.power / gain) * 1000),1000),0) with gain = 5875
You did not enter an xxxx value?

gurkc006, 2023-07-29:
Some seconds ago I typed:
python APSystem_setInverterPower.py -p 6000
which sets the output power to maximum (1000 promille) as the 6000 is bigger than the 5875).

Pmachat, 2023-07-29:
ok i understand.
So iPower = max(min(int((6000 / 5875) * 1000),1000),0)=max(min( 1000,1000),0)=1000
you put the maximum (1000 promille = 100%)
Do you know what the initial value of iPower was?

gurkc006, 2023-07-29:
On my ECU in the morning, after startup of the DS3s, the iPower is 300. Don't know why. I'm not shure, if the DS3 really have this power throttling value after startup, maybe no. Maybe it's only in the ECU. I have to check on a sunny day...

Pmachat, 2023-07-29:
I hope you have some sun soon ;-)

NOTE: Were the panels producing when you made the change on your ECU. Without sun, the DS3Ls go out
Parameter 40193 must also be at 1

gurkc006, 2023-07-31:
For you info: I checked the PV production today. In the morning, my ECU told me, that the production is throttled to 30% (300 promille) when only reading the registers from the ECU. BUT my DS3s producing unthrottled. So seems to be a "problem" of the ECU only. After setting the registers with my script, set value and real production is the same. Until the next morning.

Jeroen88, 2023-08-03:
> Sure, for anyone, who would like to control DS3 output power via ECU. You have to enable Modbus in ECU settings and assign Modbus addresses to your DS3 units. In the script, enter ONE of your Modbus [...]

How do you enable Modbus? In my ECUAPP and my EMA App there is no option to switch on Modbus, If I try a connection to port 502 (Modbus TCPIP) it is not possible. Do you know how to enable Modbus?

gurkc006, 2023-08-04:
You might try, if your ECU have a undocumented web access: Try this website in your webbrowser:
http://ip.of.your.ecu/index.php/realtimedata
or
http://ecu.fritz.box/index.php/management
Unter Management you can navigate to "Sunspec Modbus", set modbus to "on" and assign an address for each of your DS3 and click "update" afterwards. Baudrate is not necessary to set for modbusTCP.
Click "update" to activate your new settings. After that, access with my script should be possible.

sebastianrothe, 2023-08-04:
FYI: The **ECU-B** does not have this web access. Only the more expensive ones.

Jeroen88, 2023-08-04:
@gurkc006 I tried the http request but no response. What I understand from Tweakers.net is that APSystems has disabled the HTTP access in the newer firmwares. Now it is impossible to enable Modbus.

@sebastianrothe thnx for the heads up, I am aware of that. I have (or had I am going to send it back) an ECU-R with RS485 port and SunSpec Alliance logo at the bottom. But the serial number starts with 2160, and, as I understand, the HTTP access has been removed by APSystems, so no way to enable Modbus. I tried port 502, but no connection possible. I tried port 8899 and that connects, but it this port uses the APSystems proprietary protocol for ECU data read out. I do not know if and how this could be used to switch on Modbus.

JMQ1961, 2023-09-20:
I found the following  the french voltaique forum, 
the ECU-R and ECU-C have  hidden pages .

https://forum-photovoltaique.fr/viewtopic.php?f=16&t=61940&hilit=apsystems+hidden

enter the following in your webbrowser.

http://ip.of.your.ecu/index.php/hidden

stundenblume, 2023-09-21:
@Jeroen88
A reboot of the ECU helps to reach the internal webpages. It's often necessary, so a Tasmota plug is really helpful.
Tested on:
ECU-R
via WIFI
Firmware 2.1.13

Jeroen88, 2023-09-21:
> I found the following in the french voltaique forum, the ECU-R and ECU-C have hidden pages .
> 
> https://forum-photovoltaique.fr/viewtopic.php?f=16&t=61940&hilit=apsystems+hidden
> 
> enter the following in your webbrowser.
> 
> http://ip.of.your.ecu/index.php/hidden
> 
> User /PW
> 
> admin/admin

REALLY? I am going to try this, thank you! :)

Jeroen88, 2023-09-21:
> @Jeroen88 A reboot of the ECU helps to reach the internal webpages. It's often necessary, so a Tasmota plug is really helpful. Tested on: ECU-R via WIFI Firmware 2.1.13

Thank you, but I have sent the ECU-R back and bought a (second hand) ECU-C which has RS485.