patience4711/read-APSystems-YC600-QS1-DS3 #42: CC2531: starting ZB coordinator failed - failure or expected behaviour?

emuell36, 2022-08-10:
1st: this is a real great project - many thanks to patience4711 and all otheres who are contributing!

I am currently setting up my 2 panel set up (at the moment I am at that point where I have to construct a mounting for the panels),
and try to get the zigbee stuff running so everything is in place as soon as the panels are on the roof :)

Failure: starting ZB coordinator failed

Question: is this just due to the fact that I haven't configured a inverter yet, or do I have a serious problem?

Config:
- NodeMcu: https://www.amazon.de/gp/product/B09QFYHD42/
- CC2531 with CC25xxfirmware ds3\CC2530ZNP_2592-with-SBL.hex
- Currently no inverter installed

10;health answer:
[image]
[image]

patience4711, 2022-08-15:
Thanks Emuell, it's really nice to hear that sometimes. 
As for your problem i don't think you have to worry, everything seems fine to me. The inmessages are the expected answers from the zigbee module. However you should provide a correct ecu-id of 12 characters (settings). The answer then wil be something like FE0E670000FFFF80971B01A3D600000709001F which is oke.

emuell36, 2022-08-19:
I believe this is the least a user of your project could do! I would even go a step further and offer small donation since I really like how you steer the project and I am benefitting from it!

Sadly after installing my DS3 on the roof I have got trouble to pair it (CC2531 with CC2530ZNP_2592-with-SBL.hex):
[image]
[image]

emuell36, 2022-08-19:
and suddenly it worked (after about 3000 pairing attamps ;) ). Now I just have to integrate it via MQTT into the rest of my house :)
THANKS again for this GREAT project!!!

emuell36, 2022-08-19:
Now every thing is running as expected :)

closed 2022-08-19 by emuell36