patience4711/read-APSystems-YC600-QS1-DS3 #43: Thanks!

github-iw, 2022-08-17:
Thank you very much @patience4711 for this project which helped me (after some failed attempts) to read values from my QS1 and YC600 inverter. ESP8266 and zigbee network are up now and I can publish values to home assistant. As I'm not that familiair how to split the MQTT message into values per panel, can someone share his code from HomeAssistant?

code:

mqtt:
  sensor:
    - state_topic: "homeassistant/solar"
      name: "Zonnepanelen_1"
      unique_id: "solar_1"
      device_class: "energy"
      state_class: "measurement"
      unit_of_measurement: "W"
      value_template: "{{ value_json.p0 }}"

Hope I posted this in the correct place as it is not a real issue related to the softwarepackage itself

frtz13, 2022-08-19:
In H.A., I made an automation for each inverter.
```
alias: PV_prod_inv1
description: ""
trigger:
  - platform: mqtt
    topic: homeassistant/solar
    value_template: "{{ value_json.inv_serial }}"
    payload: "inv1_serialnumber"
condition: []
action:
  - service: mqtt.publish
    data:
      topic: homeassistant/solar/inv1
      payload: "{{ trigger.payload }}"
      retain: true
mode: parallel
max: 2
```
analog for inv2. you will get data from each inverter in seperate mqtt topics.

elektrinis, 2022-08-20:
This project is great.
I wonder if it would be possible (in future) to pair DS3 directly to home assistant without the ESP.

github-iw, 2022-08-21:
> alias: PV_prod_inv1
> description: ""
> trigger:
>   - platform: mqtt
>     topic: homeassistant/solar
>     value_template: "{{ value_json.inv_serial }}"
>     payload: "inv1_serialnumber"
> condition: [] [...]

Thank you very much, this was the missing solution for me to get the per panel information in home asisstant!

closed 2022-08-21 by github-iw

n000n000, 2023-02-14:
Thanks for the script : I added as an automation (my first one as I'm a newbie)
But maybe I miss something on my configuration.
If I want to use the homeassistant/solar/inv1 topic, is my Mosquitto configuration correct : 
_receive topic: homeassistant/solar
send topic: homeassistant/solar
username: user
password : passwd_

And in my Mosquitto broker in HA, I put:
In Logins:
_- username: user
  password: passwd_
In Customize: 
_active: false
folder: mosquitto_

Thanks