patience4711/read-APSystems-YC600-QS1-DS3 #47: Feature request: QT2 compatibility

HA-TB303, 2022-09-01:
First of all, great work on this privacy and freedom friendly project!

How likely is it that the upcoming QT2 will be supported? I looks like a 3-fase version of the Qt1.
https://emea.apsystems.com/nl/portfolio-item/apsystems-qt2

Hansiart2, 2022-09-06:
Wel that is a question for nostradamus. If they use the same zigbee protocol and we can find someone who has this inverter and an aps ecu and is willing to sniff and anaylize the communication....

HA-TB303, 2022-09-06:
I’m planning to buy. As soon as it’s available, i’ll volunteer. 😎

closed 2022-10-04 by patience4711

TweakM, 2023-04-19:
Any progress yet? @patience4711 : will this issue be reopened? 
I have a bunch of QT2’s and are willing to test any new software. I have a raspberry pi with a H-2 Z-002 (TICC2530+CC2591) UART to ZigBee (IEEE802.15.4) Module with 2.4G antenna (max. 400m). The firmware of the CC2530 has already been flashed to the one in this project :)