patience4711/read-APSystems-YC600-QS1-DS3 #50: Polling via MQTT does not work?

elektrinis, 2022-09-03:
Sent from another device:
`"POLL":0`

Received in console:
`mqtt received "POLL":0`

But then nothing happens.

Also in there a command to poll all inverters at once?

Hansiart2, 2022-09-05:
Correct, this does only work when automatic polling is disabled.  No such command but why not. I' look at it over 2 weeks, now on holidays

patience4711, 2022-09-19:
I assume that if you have seen the instructions in the wiki, you should have seen that the autopolling should be disabled to make this work. I changed the uppercase POLL to poll.

patience4711, 2022-09-20:
Here is a unofficial 9_5 version [esp-ecu-v9_5_BETA](https://1drv.ms/u/s!AkMG5FvUwhedi3bBwwrSfOdlptlS?e=FuWs4L)

Could you install it and check if it works as desired.

elektrinis, 2022-09-20:
Will check it next week maybe, bit overwhelmed with work now.
Thank you!

closed 2022-09-21 by patience4711