patience4711/read-APSystems-YC600-QS1-DS3 #51: Any serial will pair

elektrinis, 2022-09-04:
I've mistyped my serial number, and yet it "paired". But then I was getting error 11 when trying to poll.

feature request: any way to get zigbee signal strength? Two of my inverters "paired", but I was unable to poll them (error 11) until I brought the radio closer and re-paired.

Hansiart2, 2022-09-05:
I' don' t remeber i have seen a zigbee request for signal strength and that this is implemented in the firmware so i think that the answer is no. I wll look at it anyway and come back when found something.

elektrinis, 2022-09-05:
Signal strength is being reported by standard firmware used for smart home integrations via zigbee2mqtt (in home assistant, etc). Not sure about this one though.
Nevertheless I think there is something wrong with pairing process, because it will pair with any serial if there are working inverters close by. It does not paid during the night when inverters are off.

togoright, 2022-09-05:
[image]

FE 72 44 81 0000 0601 7434 14 14 00 **_AD_** 00 952C09 00 00 ...

**_AD_** is the LinkQuality in the example. 00 is bad, FF is good.

I display it as a percentage with the following formula.
AD = 173 *100/254 = 68%

Hansiart2, 2022-09-06:
I don' t think that the pairing is a big problem but if it is, it would be a design error of APS. Nothing we can do about that i am afraid. I agree that this is strange, in the communication the serial nr is checked so when we send a request to serial x and serial y answers with its own seralnr, this shouldn't be considered as a valid pairing. Anyway the inverter can't be polled so it is not actually paired 
i would think. 

About the signal strength you found the answer already. Good work. Maybe i can make a console command for this.

togoright, 2022-09-19:
Hello,

the Link Quality is defined in the IEEE 802.15.4. I find this information [here](https://de.wikipedia.org/wiki/Link_Quality_Indicator).
Other than that, I haven't found any coherent information, all pretty scattered.
Like here:
https://github.com/Koenkk/zigbee2mqtt/issues/504#issuecomment-431550198
https://tasmota.github.io/docs/Zigbee/#reading-sensors   (a little deeper under Supported values: )

I use the formula because I find it easier to estimate the value that way.

I also did a few tests at different distances. Likewise, I could see that the further away I was, the worse the value was, the closer I was to the inverter, the values kept getting better.

patience4711, 2022-09-19:
Hi @togoright
Thanks for your contribution. Is is clear that the value varies with the distance/obstructions. And the highest value represents the best quality.  Its a good idea to convert this to a percentage.

In theory the highest value can be 0xdd which is 255 * 100 / 255 = 100%
The lowest = 0x00 = 0 * 100 / 255  = 0%
The middle value = 127,5 = 0x7d = 127,5 * 100 / 255 = 50 (it does not have this precission)
So shouldn't the formula not be *100/255 ?  It doesn't matter much but i was wondering why you choosed 254.

togoright, 2022-09-20:
Excuse me, of course 255 is the correct divisor. It's a typo. 😒

closed 2022-09-21 by patience4711