patience4711/read-APSystems-YC600-QS1-DS3 #54: Compiling on Arduino

elektrinis, 2022-09-20:
I wanted to change some things (like polling rate) so tried to compile it, but there were many issues.
At first, many libraries were missing. I have found some (like SunMoon) on the internet, but not sure if those are correct ones, also what versions should be use.
Anyway I have managed to get it compiling, and it did without errors, however my ESP8266 crashes right after WiFi connection is attempted.
I'm not Arduino guru and in need of some assistance :(
Can you check what versions of libraries should be used, when where the non-standard ones should be downloaded from?
I think it would be great if this could be specified in the source code next to include.
Also I am not sure that setting should be set in Arduino's dropdown menus, like memory type etc.

patience4711, 2022-09-20:
When i published the source code i said that i would not support people who want to do their own compiling. This is all too time consuming for me. As for the ide settings see the picture below. 
When you intensify the polling rate you ask for stability problems. 
I don't think that the crashes have something to do with the version of the libraries, in most cases these are the latest versions.

[image]

elektrinis, 2022-09-20:
Thank you! Will try.

closed 2022-09-27 by patience4711