patience4711/read-APSystems-YC600-QS1-DS3 #56: Error polling failed 15 with DS3-S v9.5

baba2k, 2022-09-21:
Hi all, i wanna use an APSystems DS3-S with this great tool. Pairing was successfull but the polling fails with error code 15. 

```
* * connection opened * *
set diagnose to 1
zb send poll cmd inverter 0 cmd:1D2401146C1414060001000F1380971B01A3D8FBFB06BB000000000000C1FEFE51
polling answer FE0164010064FE034480001401D2
no route receipt
ignoring, received FE0164010064FE034480001401D2
polling failed 15

```

Any idea?

Thanks!

baba2k, 2022-09-21:
My Hardeware is NodeMCU + [H2 Z-002 (TICC2530 CC2591)](https://www.amazon.de/dp/B01FXC1G5M). My inverter ist an APSystems DS3-S and the Serial ist correct. The pairing was successfully, only the polling does not work. The Hardware is only 4m away from the inverter.

baba2k, 2022-09-21:
But i'am using an 1A/3W (5V) power supply for the NodeMCU, is this enough? At the wiki it is sometimes 3W and sometimes 3A

patience4711, 2022-09-21:
If the pairing works, the power is enough is would think. 
You are using a brand new version that i have not tested, i dont have currently the hardware available. There is a small chance that this could be the cause.
You could install the previous version  [9_4](https://1drv.ms/u/s!AkMG5FvUwhedi0bA08s8itWhnSJA?e=2ebtEf) using the sw update option in the menu, Let me know if this has any result.

baba2k, 2022-09-21:
I tried it again and it works now! This morning there was little sun, maybe it was too little? The DS3 has according to the LED also produced power this morning

patience4711, 2022-09-21:
Oke good, sometimes it needs a little time. 
So you are still working with version 9_5?  Do you see the signal quality value in the info page? Can you tell me what the value is?

baba2k, 2022-09-21:
Yes it is 9.5 and signal strength is 84.4%

patience4711, 2022-09-21:
That looks good. If you have more data as to the distance ecu - inverter and the signal quality value, you can help me greatly by reporting them.  With your module i would be interested in the value at very close range and the value at which the polling answers fail. This way i can compare different zigbee modules.
Thanks in advance

baba2k, 2022-09-25:
Sure! Today i've installed my panels. First test: 5 meter with wall 57,3% and no answers failed. I will do some more tests in the next days and give you the results.

@patience4711 I miss the current and voltage of the panels and ac volt + signal quality in the mqtt json message. Also temp, p0 and p1, energy are strings and not float/numbers. Ist it possible to fix this in a future version? The panels are numbered from 1 on the inverter and in the web interface, but in the MQTT JSON message the panels start with 0.

[image]

Thanks!

patience4711, 2022-09-26:
You want all the data, the json would be something like this.
{"serial":"x","acv":x,"freq":x,"temp":x,"dcv":[dcv0, dcv1, dcv2, dcv3],"dcc":[dcc0,dcc1,dcc2,dcc3],"pwr":[p0, pw1, pw2, pw3], "en":[en0, en1, en2, en3]}
Would that be what you want? Forget the numbering on the inverter, in software the 0 comes first.
In this test version you can enable mqtt by choosing a format. 
format 3 is what you want as far as programming space this allows. I left the totals out because your software can calculate this easely.
[ESP_ECU-v96_BETA](https://1drv.ms/u/s!AkMG5FvUwhedi3nSCaTzfI6882Mf?e=kGtNz4)
Could you test this and report back?

baba2k, 2022-09-26:
Great, that was fast :-) Thank you! I already installed it and will give you feedback tomorrow. What do you think of a "total energy" in addition or instead of "energy per day"? The Shelly Pro 1 PM does it that way for example. Currently, data is lost once the home automation system is down for one day. Also, it is not easy for me to add up the individual days to calculate a "total energy" myself. The home automation system expects a counter variable that does not reset itself every day.

baba2k, 2022-09-26:
_The total energy you mean, is that the total over multiple days? That is not very usefull because that value would be lost at every reset._

Didn't saw this. Okay then i've to find a solution in my home automation system.

baba2k, 2022-09-28:
Great, it works! Is it possible to construct the json separated by channel? Currently, the representation in the home automation system is not grouped by panel:
[image]

I would need a json construct like this to get a separation by channel/panel:
`{
  "serial": "aaa",
  "acv": 233.3,
  "freq": 50.2,
  "temp": 18.2,
  "ch0": {
    "dcv": 35.1,
    "dcc": 0.9,
    "pwr": 33.8,
    "en": 112.16
  },
  "ch1": {
    "dcv": 35.2,
    "dcc": 0.7,
    "pwr": 31.3,
    "en": 124.18
  },
  "total": {
    "pwr": 62.5,
    "en": 413.29
  }
}`

Is this possible with not too much effort? Thank you very much!

I have seen that the "energy per day" remains after all over a reboot. Doesn't it make sense to create a "total engery over multiple days" like the other power meters (and ECU-R) have? There could be a field in the web interface where the data can be set or restored manually after a reset.

Next range test over 24 hours: 8 meters, 2 walls and 21.2% signal quality.  Everything works great!

patience4711, 2022-09-28:
That would be a long string with dcv dcc pwr en 4 times in it. What would you think of this much more efficient json.? This way you have to remember yourself "dcv dcc pwr en", your software can handle that.
But i think your software should be able to group the data any way you want from the mqtt message. 
I don't have much more programming space left to fullfill every individual wishes as to the mosquitto format.
maybe i can add this:
}
  "serial": "aaa",
  "acv": 233.3,
  "freq": 50.2,
  "temp": 18.2,
  "ch0": [35.1, 0.9, 33.8, 112.16],
  "ch1": [35.2, 0.7, 31.3, 124.18],
    "totals": [62.5,413.29]
 }

baba2k, 2022-09-29:
Interesting, I didn't know that the length of the JSON string is relevant. Then gladly also in this format. However, the new format in 9.6 also looks good: https://github.com/patience4711/read-APSystems-YC600-QS1-DS3/issues/59#issuecomment-1262475268

patience4711, 2022-09-29:
That is not an mqtt message but the answer to a htpp request. The longer the string the more programming is involved, this is relevant since programming memory space is almost consumed. I like to keep some spare memory for really important issues. That is why i skipped the signal quality, that is not a key figure. In the end the only key figure is yield i think ...

baba2k, 2022-09-30:
@patience4711 the first array element is always 0 at "totals": [62.5,413.29]

Thank you!

patience4711, 2022-09-30:
Yes i forgot that i left out totals assuming that home automation software can totalize values.  If you download and install again (vers 13:00hr) it should be fixed. I can't test myself so this is all a bit difficult.

closed 2022-10-04 by patience4711