patience4711/read-APSystems-YC600-QS1-DS3 #57: Home Assistant: settings

markvd1962, 2022-09-24:
Thanks @patience4711, works really perfect (2xYC600, 1x QS1). What settings to use in the mosquitto section of the ESP-ECU program so values are published to be ready to pickup by home assistant? Tried everything.

frtz13, 2022-09-24:
you may want to have a look at issue #43

markvd1962, 2022-09-24:
I saw #43 and tried to implement this is Home Assisitant. But I mean the setting in ESP ECU, I think it goes wrong here. It has a tab "mosquitto". What should I use for "receive topic" and "send topic". And should I enter password and user id for my mosquitto broker in Home Assistant?

se3, 2022-09-25:
> And should I enter password and user id for my mosquitto broker in Home Assistant?

Yes, username and password is for mosquitto broker.
>What should I use for "receive topic" and "send topic"?

For topic you can use what ever **you** want to receive topic on Home Assistant side, since you configure topic manually on sender and receiver side as foo/bar.
In #43  "homeassistant/solar" is used.
For better reading you could use:
receive topic: house/ecu or homeassistant/ecu
send topic: house/inverter or homeassistant/inverter

Check out [Understanding MQTT Topics](http://www.steves-internet-guide.com/understanding-mqtt-topics/)

markvd1962, 2022-09-25:
Exactly what I needed for now. Thanks for the great support. I succeeded in getting a reading in Home Assistant. It is for one panel only and for one inverter only. But I can clearly see it in HA by using the developer tools. Also thanks for the link to "understanding MQTT Topics". I am going to spend time to read, hopefully understand and get reading of all my panels (over the past year I learned a lot, but so many more to learn).

se3, 2022-09-25:
hi @markvd1962,
more information about Home Assistant (HA) will come soon. 
I'm sure we will help @patience4711  to put all working solution into Wiki.

At the moment in Wiki the json format description is missing (only domoticz is present).
Here the information I got from source code and MOSQUITTO HELP window:

MQTT poll message receved in Home assistant MQTT client listener (see mqtt sensor)

```
"homeassistant/inverter" : // send topic: from "mosquitto configuration" window
{
    "inv_serial": "<number>", // 12 digits serial number
    "temp" : "<number>",  // temperature in °C ?
    "p0" : "<number>",  // power in Watt?
    "p1" : "<number>",  // power in Watt?
    "p2" : "<number>",  // power in Watt?
    "p3" : "<number>",  // power in Watt?
    "energy" : "<number>" // Wh?
}
```
possible HA configuration (**! not tested !**):

**home assistant automation:**
Since we receive mqtt message from one ECU (topic) which can handle more than one inverter we have to create some rules and sensors on HA side for every solar panel.
First as @frtz13 mentioned we pass ecu mqtt message to correspondig single inverter topic depend on inverter serial number:
```
alias: ECU inverter multiplexer
trigger:
  - platform: mqtt
    topic: homeassistant/inverter
    value_template: "{{ value_json.inv_serial }}"
    payload: "put here 12 digits serialnumber of your inverter 1"
condition: []
action:
  - service: mqtt.publish
    data:
      topic: 'homeassistant/inverter/1'
      payload: '{{ trigger.payload }}'
      retain: true
mode: parallel
max: 2
```

Create now Panel Sensors for each pannel
configuration.json:
```
mqtt:
  sensor:
  - state_topic: "homeassistant/inverter/1"
    name: "Inverter 1 Energy Total"
    unique_id: "inverter1_energy"
    device_class: "energy"
    state_class: "measurement"
    unit_of_measurement: "Wh"
    value_template: "{{ value_json.energy }}"
  - state_topic: "homeassistant/inverter/1"
    name: "Inverter 1 Panel 1"
    unique_id: "inverter1_p0"
    device_class: "energy"
    state_class: "measurement"
    unit_of_measurement: "W"
    value_template: "{{ value_json.p0 }}"
  - state_topic: "homeassistant/inverter/1"
    name: "Inverter 1 Panel 2"
    unique_id: "inverter1_p1"
    device_class: "energy"
    state_class: "measurement"
    unit_of_measurement: "W"
    value_template: "{{ value_json.p1 }}"
  - state_topic: "homeassistant/inverter/1"
    name: "Inverter 1 Panel 3"
    unique_id: "inverter1_p2"
    device_class: "energy"
    state_class: "measurement"
    unit_of_measurement: "W"
    value_template: "{{ value_json.p2 }}"
  - state_topic: "homeassistant/inverter/1"
    name: "Inverter 1 Panel 4"
    unique_id: "inverter1_p3"
    device_class: "energy"
    state_class: "measurement"
    unit_of_measurement: "W"
    value_template: "{{ value_json.p3 }}"
```

Please note this is not tested example based on collected information in Wiki, issues, and source code...
Once I get DS3D paired (hope on more sunny days) I will publish all examples.
/se3

markvd1962, 2022-09-26:
Hello se3,
Thanks for all the information. This will keep me busy the next few days/weeks (software is my weak spot, but makes fun). And it is getting dark very quicky after getting back home from work. Keep you posted after a breakthrough.
/mark

frtz13, 2022-09-26:
@se3 for the energy sensors, I'd rather use:
```
      - name: "Inverter 1 Energy Total"
        device_class: energy
        state_topic: "homeassistant/inverter/1"
        value_template: '{{ ((value_json["energy"] | float(0)) / 1000) | round(3) }}'
        unit_of_measurement: "kWh"
        state_class: "total_increasing"
```
for the power sensors:
```
      - name: "Inverter 1 Panel 1"
        device_class: power
        state_topic: "homeassistant/inverter/1"
        value_template: '{{ value_json["p0"] }}'
        unit_of_measurement: "W"
        expire_after: 330
```
the expire_after option will give you an 'unavailable' state for the power sensors when zigbee controller needs some time for reset.

se3, 2022-09-27:
Just to mention...
Home Assistant support [MQTT Discovery](https://www.home-assistant.io/docs/mqtt/discovery) feature that is included also in [Tasmota integrations](https://tasmota.github.io/docs/Home-Assistant/) but need to be enabled by "SetOption19 0".
With this feature there is no need to configure sensor on Home Asistant side, this could be done by ECU using e.g. following MQTT command:

Example create sensor "Inverter 1 Energy Total":

mosquitto_pub -r -h 127.0.0.1 -p 1883 -u <mosquitto username> -P <mosquitto password> -t "homeassistant/sensor/inverter0energy/config" -m '{"name": "Inverter 0 Energy Total", "device_class": "energy", "state_topic": "homeassistant/sensor/inverter0energy/state", "unit_of_measurement": "kWh", "value_template": "{{ value_json.energy}}" }'

In this case @patience4711 could have full control on the structure of sensors in case he want change json structure or add some more values / sensors.

patience4711, 2022-09-27:
Just a remark. I am testing with an extra json format which looks like
{"serial":"x","acv":x,"freq":x,"temp":11.8,"dcv":[39.2,39.2,0.0,0.0],"dcc":[0.0,0.0,0.0,0.0],"pw":[0.4,1.0,0.0,0.0],"en":[3.9,3.3,0.0,0.0]}
I am not thinking of a dedicated home assistant format. This is a pretty common format that can be handled by most of the domotica systems. There is not much programming space anymore that i can sacrifice for things like this, without the risk that ota gets impossible.

markvd1962, 2022-09-28:
Quick feedback: I used the code of se3 with the suggestions of frtz13 included. It all works! Really great. Thanks to all of you for your support. I am going to enjoy my APS solar panel data in HA.

patience4711, 2022-10-06:
The integration with all kinds of home automation systems is not part of the wiki because i think that it doesn't belong there. For me it belongs to HA where it is (or should be) described how to setup a sensor. I don't know about ha, timberwolf, domoticz and whatever more automation systems so I'll stick to sending the data and nothing more.

closed 2022-10-06 by patience4711

babajun12, 2022-11-23:
I used the HA integration the same style as se3 from above which worked fine.
Since the upgrade to v9_6 with the different format, I have no clue how to exctract the string for AC, DC... out of format4.
Does somebody have a hint for me how to integrate?
Thx

frtz13, 2022-11-23:
I do not use that format, but it should work like this for voltage and current of first panel of inv1
```
mqtt:
    sensor:
      - name: "Inv1 Panel1 Volt"
        state_topic: "yourtopic/inv1"
        value_template: '{{ value_json["ch0"][0] | float(0) }}'
        unit_of_measurement: 'V'
        device_class: voltage
        expire_after: 330
      - name: "Inv1 Panel1 Curr"
        state_topic: "yourtopic/inv1"
        value_template: '{{ value_json["ch0"][1] | float(0) }}'
        unit_of_measurement: 'A'
        device_class: current
        expire_after: 330
```
etc.

h4nk0r, 2023-02-01:
It´s also possible without mqtt, example below with the usage of rest

`
sensor:
  - platform: rest
    name: "ecu_esp"
    resource: http://ip_ecu_eps/get.Inverter?inv=0
    scan_interval: 60
    value_template: "OK"
    json_attributes:
      - serial
      - freq
      - temp
      - avc
      - dcv0
      - dcc0
      - pow0
      - en0
      - dcv1
      - dcc1
      - pow1
      - en1
      - power
  -  platform: template
     sensors:
      ecu_esp_serial:
        value_template: "{{ state_attr('sensor.ecu_esp', 'serial') }}"
        friendly_name: "APSYTEMS Seriennummer"
      ecu_esp_freq:
        value_template: "{{ state_attr('sensor.ecu_esp', 'freq') }}"
        friendly_name: "APSYSTEMS AC Frequenz"
      ecu_esp_avc:
        value_template: "{{ state_attr('sensor.ecu_esp', 'avc') }}"
        friendly_name: "APSYTEMS AC Spannung"
      ecu_esp_dcv0:
        value_template: "{{ state_attr('sensor.ecu_esp', 'dcv0') }}"
        device_class: voltage
        unit_of_measurement: 'V'
        friendly_name: "APSYTEMS Panel 0 DC Spannung"
      ecu_esp_dcc0:
        value_template: "{{ state_attr('sensor.ecu_esp', 'dcc0') }}"
        unit_of_measurement: 'A'
        friendly_name: "APSYTEMS Panel 0 DC Stromstaerke"
      ecu_esp_pow0:
        value_template: "{{ state_attr('sensor.ecu_esp', 'pow0') }}"
        device_class: power
        unit_of_measurement: 'W'
        friendly_name: "APSYTEMS Panel 0 Leistung"
      ecu_esp_en0:
        value_template: "{{ state_attr('sensor.ecu_esp', 'en0') }}"
        device_class: power
        unit_of_measurement: 'W'
        friendly_name: "APSYTEMS Panel 0 Leistung Heute"
      ecu_esp_dcv1:
        value_template: "{{ state_attr('sensor.ecu_esp', 'dcv1') }}"
        device_class: voltage
        unit_of_measurement: 'V'
        friendly_name: "APSYTEMS Panel 1 DC Spannung"
      ecu_esp_dcc1:
        value_template: "{{ state_attr('sensor.ecu_esp', 'dcc1') }}"
        unit_of_measurement: 'A'
        friendly_name: "APSYTEMS Panel 1 DC Stromstaerke"
      ecu_esp_pow1:
        value_template: "{{ state_attr('sensor.ecu_esp', 'pow1') }}"
        device_class: power
        unit_of_measurement: 'W'
        friendly_name: "APSYTEMS Panel 1 Leistung"
      ecu_esp_en1:
        value_template: "{{ state_attr('sensor.ecu_esp', 'en1') }}"
        device_class: power
        unit_of_measurement: 'W'
        friendly_name: "APSYTEMS Panel 1 Leistung Heute"
      ecu_esp_power:
        value_template: "{{ state_attr('sensor.ecu_esp', 'power') }}"
        device_class: power
        unit_of_measurement: 'W'
        friendly_name: "APSYTEMS Gesamt DC Leistung"

`

h4nk0r, 2023-02-01:
> And will this work with more than 1 inverter? Or do I have to copy the whole code for all inverters?

For each inverter extra because of

>  resource: http://ip_ecu_eps/get.Inverter?inv=0

swbouman, 2023-02-01:
Whoops sorry! 😇

super job!

n000n000, 2023-02-14:
> 

Hello h4nk0r
I've tried your rest sensor script but I get this message when I try to get the infos : 
_This entity ('sensor.ecu_esp_freq') does not have a unique ID, therefore its settings cannot be managed from the UI._
Any idea why ?

h4nk0r, 2023-02-14:
Hey n000n000,

try this one, Maybe add it to the other ones with a separate name

ecu_esp_freq:
  value_template: "{{ state_attr('sensor.ecu_esp', 'freq') }}"
  friendly_name: "APSYSTEMS AC Frequenz"
  unique_id: "APSYSTEMS_AC_Frequenz"

Greetings
h4nk0r

Von: n000n000
Gesendet: Dienstag, 14. Februar 2023 17:37
An: patience4711/read-APSystems-YC600-QS1-DS3
Cc: h4nk0r; Comment
Betreff: Re: [patience4711/read-APSystems-YC600-QS1-DS3] Home Assistant:settings (Issue #57)

Hello h4nk0r
I've tried your rest sensor script but I get this message when I try to get the infos :
This entity ('sensor.ecu_esp_freq') does not have a unique ID, therefore its settings cannot be managed from the UI.
Any idea why ?

n000n000, 2023-02-15:
Thank you h4nk0r : it's working perfectnow
I even added some calculations to see my overall production / instant production.

... and thank a lot to patience4711 for his projet.

kokobadu, 2023-11-15:
used that for some time, worked great. Tnx for that.
Now format is changed I belive:
{"inv":2,"name":"three","polled":true,"type":1,"serial":"8060000xxx","sid":"25E3","freq":50,"temp":11.3,"acv":227.8,"sq":30.6,"pw_total":27.5,"en_total":0.22,"dcv":[28.8,29.1,29.4,29.1],"dcc":[0.2,0.3,0.2,0.2],"pow":[6.1,9.7,5.2,6.5],"en":[56.09,68.45,40.26,59.26]}

Does the sensor has to look like this:
json_attributes:
    - dcv[0]
    - dcc[0]
    - pow[0]
    - en[0]
    - dcv[1]
    ... ect...
    ??
is not working like that.. 
sorry if the q is stupid, got lost.