patience4711/read-APSystems-YC600-QS1-DS3 #58: Pairing DS3-D inverter fails

se3, 2022-09-25:
Hi patience4711,

Thank you for sharing the project with us.

I owned an [DS3D](https://emea.apsystems.com/wp-content/uploads/2022/09/4303604202_APsystems-Microinverter-DS3D-For-EMEA-Datasheet_-Rev1.1_2022-08-19.pdf) inverter connected 4 pannels to it with 2 pannels to each MPPT input.
The system is not connected to 230V yet.

Hardware I use: Wemos D1 mini + CC2530.
[image]

Flahed ESP-ECU_v9_5.bin (ESP8266) and CC2530ZNP-with-SBL.hex firmware (CC2530)  (I was not sure about 2591 or 2592).
Connection to Zigbee device seems to be OK as you see in the log.
During paaring I get polling failed 15 which means not enough sun?

**Console**
```
* * connection opened * *
--
set diagnose to 1
check zb system
inMessage = FE02610179071C rc = 7
ping ok
inMessage = FE0E670000FFFF80971B01A3D8000007090011 rc = 19
found 0709, running oke
poll inverter 0
zb send poll cmd inverter 0 cmd:1D240100001414060001000F1380971B01A3D8FBFB06BB000000000000C1FEFE29
polling answer FE0164010064FE034480001401D2FE274481000006000000141400FF0000000000001380971B01A3D8FBFB06BB000000000000C1FEFE00000002
no route receipt
ignoring, received FE0164010064FE034480001401D2FE274481000006000000141400FF0000000000001380971B01A3D8FBFB06BB000000000000C1FEFE00000002
polling failed 15
```
**System log**
```
Time          Type	Command
25-13:51:35 | pair    | failed, inverter got id 0x0000
25-13:51:18 | zigbee  | zigbee module reset
25-13:51:18 | pairing | start pairing inverter sn 70600004xxxx
25-13:50:32 | zigbee  | coordinator up
25-13:50:31 | zigbee  | initial healthcheck
25-13:50:26 | mqtt    | connected
25-13:50:25 | system  | boot up
25-13:50:25 | system  | time retrieved
```

Is DS3-D supported by the software (decoding)?
How to add 4 pannels to one inverter?

If you need more debug informtion about DS3D I can provide more info.
Hopefully we will have more sun next days :)

Thank you
/se3

IerwinDE, 2022-09-25:
I am in a similar situation and can not get the device to pair with my DS3-D.
You mentioned that yours is not yet connected to 230V. Does that make any difference or is it even a requirement for pairing?
Becaue mine is connected and currently generating over 300 Watts but still does not seem to want to talk to me.

Hope that @patience4711 will chime in and tell us what we can do to help get the DS3D integrated.

IerwinDE, 2022-09-25:
Heres the log from my pairing attempt:

`
inMessage = FE00660066FE0145C0088CFE0145C0088CFE0145C0088C rc = 23
trying pair inv 0
pair command = 2524020FFFFFFFFFFFFFFFFF14FFFF140D0200000F1100706000049001FFFF10FFFF80971B01A3D8FE
sending paircmd 2
received : FE0164020067FE0145C0088CFE0145C0088CFE0145C0088CFE0145C0088CFE0145C0088CFE0145C0088C
pair command = 1A24020FFFFFFFFFFFFFFFFF14FFFF140C0201000F0600706000049001B0
sending paircmd 3
received : FE0164020067FE0145C0098D
not found serialnr, returning
pair command = 2524020FFFFFFFFFFFFFFFFF14FFFF140F0102000F1100706000049xxxA3D810FFFF80971B01A3D886
sending paircmd 4
received : FE0164020067
not found serialnr, returning
pair command = 1A24020FFFFFFFFFFFFFFFFF14FFFF14010103000F060080971B01A3D84F
sending paircmd 5
received : FE0164020067
failed, inverter got id 0x0000
sending NO cmd
inMessage = FE0164010064FE034480001401D2 rc = 14
inMessage = FE0E670000FFFF80971B01A3D8000007090011 rc = 19
found 0709, running oke`

IerwinDE, 2022-09-26:
You will not be able to get values for all 4 Panels seperately because the DS3D doesnt even know how many Panels are connected in series to each tracker

se3, 2022-09-26:
thank you @IerwinDE, now the picture is more clear for me. I will call it "MPPT tracker" in home assistant

closed 2022-09-26 by se3