patience4711/read-APSystems-YC600-QS1-DS3 #59: Problematic number format

psxde, 2022-09-29:
Hi, Telegraf cannot parse the values receiving from http://IP/get.Inverter?inv=0  resulting in "invalid character '0' after object key:value pair". It seems that the values with double and triple leading zeros like '"pow2":000.0' are the problem.

Thanks,
Patrick

patience4711, 2022-09-29:
Thank you for your interest in my software. What esp-ecu version? What do you suggest? As of version 9_6 the returned json looks like this:
{"serial":"408000158211","freq":00.0,"temp":000.0,"avc":000.0,"dcv0":00.0,"dcc0":00.0,"pow0":0.0,"en0":0.00,"dcv1":00.0,"dcc1":00.0,"pow1":0.0,"en1":0.00,"power":0.0,"energy":0.00} which means that the values are floats.

psxde, 2022-09-29:
Hi, I have updated to ESP-ECU_v9_6 but there are some values left with double leading zeros:

`{"serial":"...","freq":50.0,"temp":12.7,"avc":231.1,"dcv0":37.1,"dcc0":0.0,"pow0":0.0,"en0":597.78,"dcv1":9.9,"dcc1":0.0,"pow1":0.0,"en1":0.61,"dcv2":00.0,"dcc2":00.0,"pow2":0.0,"en2":0.00,"dcv3":00.0,"dcc3":00.0,"pow3":0.0,"en3":0.00,"power":0.0,"energy":0.00}`