patience4711/read-APSystems-YC600-QS1-DS3 #60: HomeAssitant mqtt support

JochenKlenk, 2022-10-05:
Hi guys,

would be amazing to have an right HomeAssitant Support.
I mean an full automated mqtt integration in HA.

Or do someone have an template for HA on this.

Thanks

se3, 2022-10-05:
HA config is actually missing in wiki, you can try this configuration:
https://github.com/patience4711/read-APSystems-YC600-QS1-DS3/issues/57#issuecomment-1257253196
/se3

patience4711, 2022-10-06:
This is a dupplication of a question already asked. The integration with all kinds of home automation systems is not part of the wiki because i think that it doesn't belong there. For me it belongs to HA where it is (or should be) described how to setup a sensor. I don't know about ha, timberwolf, domoticz and whatever more automation systems so I'll stick to sending the data and nothing more.

closed 2022-10-17 by patience4711