patience4711/read-APSystems-YC600-QS1-DS3 #61: Can't pair DS3-S

Viermusketiere, 2022-10-11:
I am trying to pair to an DS3-S using an Wemos D1 Mini and an CC2530.

When trying to pair I get the following output in the system console:
```
init zb coordinator
zigbee module reset
zb send cmd 0
inMessage = FE0166050062 rc = 6
zb send cmd 1
inMessage = FE064180020202020702C2 rc = 11
zb send cmd 2
inMessage = FE0166050062 rc = 6
zb send cmd 3
inMessage = FE0166050062 rc = 6
zb send cmd 4
inMessage = FE0166050062 rc = 6
zb send cmd 5
inMessage = FE0166050062 rc = 6
zb send cmd 6
inMessage = FE0164000065 rc = 6
zb send cmd 7
inMessage = FE00660066FE0145C0088CFE0145C0088CFE0145C0088C rc = 23
trying pair inv 0
pair command = 2524020FFFFFFFFFFFFFFFFF14FFFF140D0200000F1100702000030913FFFF10FFFF80971B01A3D832
sending paircmd 2
received : FE0164020067FE0145C0088CFE0145C0088CFE0145C0088CFE0145C0088CFE0145C0098D
pair command = 1A24020FFFFFFFFFFFFFFFFF14FFFF140C0201000F06007020000309137C
sending paircmd 3
received : FE0164020067
not found serialnr, returning
pair command = 2524020FFFFFFFFFFFFFFFFF14FFFF140F0102000F1100702000030913A3D810FFFF80971B01A3D84A
sending paircmd 4
received : FE0164020067
not found serialnr, returning
pair command = 1A24020FFFFFFFFFFFFFFFFF14FFFF14010103000F060080971B01A3D84F
sending paircmd 5
received : FE0164020067
failed, inverter got id 0x0000
sending NO cmd
inMessage = FE0164010064FE034480001401D2 rc = 14
inMessage = FE0E670000FFFF80971B01A3D8000007090011 rc = 19
found 0709, running oke
```

I have tried it with various distances between the CC2530 and the inverter ranging from 10m to 0.2m direct line of sight and with various objects blocking direct line of sight to exclude overpowering one end of the RF-Components.

System Info:
```
system time = 17:04 hr.   summertime
firmware version : ESP-ECU_v9_6
time retrieved today : yes
the signalstrength of the connection to the router = -39
mosquitto not configured
check the mosquitto settings
system up time: 0 days 0 hrs 36 min.
current errorCode = 10

ESP INFORMATION
--
ESP CHIP ID nr: | 14155710 | Flash Chip ID nr: | 1458208
IDE Flash size: | 4194304 bytes | Real Flash size | 4194304 bytes
Free memory | 15608 bytes | remote IP | 192.168.178.107

variables dump
value=0 inverterCount=1 zigbeeUp=1
pollRes= 300 now=1665507840 switchonTime=1665474683 switchoffTime=1665514069
unixtime=1665507840
polled = 000000000
resetCounter = 0 calliBration = 1.66 pollOffset = 0 Mqtt_Format = 0
Content filesystem :
Inv_Prop0.str size 48
basisconfig.json size 103
mqttconfig.json size 183
wificonfig.json size 104
```

If someone has an idea on how to fix the problem i would be happy if you'd like to share it with me.

Viermusketiere, 2022-10-13:
I tried it at different days, so the inverter should have been reseted multiple times. This is definitely the serial number which is noted on the inverter. When i tried to pair the sun was shining and the inverter was producing >200W measured by a smart meter. I am running out of ides... Is their a way to sniff the inverters SN via zigbee to confirm that the right one is noted on the inverter?

patience4711, 2022-10-13:
I examened the pairing commands and everything seems oke but the inverter does not answer as expected.  I don't know why but it could be a wrong serial.
At the 2nd pairing command you get 5 answers like :
FE0164020067  FE0145C0088C  FE0145C0088C  FE0145C0088C  FE0145C0088C  FE0145C0098D
which is a status command meaning that the inverter is started as Zigbee Coordinator. This is unexpected. The normal answer should be FE0164020067 which means af data request success. [you can find the meaning of zigbee commands here](https://radiocrafts.com/uploads/CC2530ZNP%20Interface%20Specification.pdf)

Maybe you can try following: change the ecu serial nr briefly, for instance from D6A3011B9780 to D7A3011B9780
, save. Check that the coordinator is running oke (healthcheck) and try pairing again.

There is a way to sniff the  zigbee trafic, the inverter sends a beacon signal with it's serial in it, if you can catch that you will be sure. You should put sniffing firmware on your zigbee module and connect it to usb of a pc/laptop, so i think you need a cc2531. There is enough to find on this subject.

joffito, 2022-10-17:
is the inverter paired with an official ECU or other device?