patience4711/read-APSystems-YC600-QS1-DS3 #62: NodeMCU doesn't show webpage

joffito, 2022-10-16:
Another bloody beginner here.

I followed the steps to flash the ZigBee module and it worked. Then moved on to the ESP and flashed the firmware and entered the wifi credentials after connecting to it. (NodeMCU AMICA ES8266)
Now i try to access it with the IP my router tells me but it's not possible to connect to it.

I tried to re-flash it (and flashing the firmware works) but it won't light up the blue LED constantly and won't open an AP so I can't connect to it again to re-enter credentials etc.
Any ideas how to fully put it back to factory mode so it will show me an AP mode just like the first time i configured it ?

joffito, 2022-10-16:
SORRY! Too stupid to read that i need to hold down the flash button to reset the wifi settings.

Anyhow, after resetting and re-configuring again the same: i can't connect to the ESP via its IP address.After power on and a brief flash of the LED it will flash 3 times and then every 3 seconds for some time and then the LED goes dark.

joffito, 2022-10-16:
ok, now after some tries i figured it out. it now showed me the correct IP and the user is admin and the pw is what you selected. maybe that's worth putting somewhere in the WIKI :)

patience4711, 2022-10-17:
It's not completely clear what extra info should be added to the wiki. Please read the chapter connect to wifi and make your suggestions. I think your initial problem was that you used an esp that had already wifi settings. In the hardware chapter it is advised to use a memory-cleaned or a new esp.

joffito, 2022-10-17:
I currently have an ECU-B in my home network. I mixed up the IP addresses of the ESP and the ECU (which seems to use an ESP as well). Then i did not know which credentials to use (username is "admin" and password obviously which was chosen during WIFI setup of the ESP). I will make a suggestion to the WIKI.
Now, after i disconnected the ECU-B i was able to successfully pair to the DS3-S inverter - great :)

closed 2022-10-18 by patience4711