patience4711/read-APSystems-YC600-QS1-DS3 #63: objcopy CC2530ZNP-with-SBL.hex fails

silbor, 2022-10-17:
Hi there, 

I am trying to convert ESP-ECU-9_6/CC25xxfirmware ds3/CC2530ZNP-with-SBL.hex to a bin to flash to a regular 2530 board, using ccloader on a D1. 

Using the instructions on https://zigbee.blakadder.com/flashing_ccloader.html I tried first on Linux (Ubuntu), then on Windows 10 with 
`objcopy(.exe) --gap-fill 0xFF --pad-to 0x040000 -I ihex CC2530ZNP-with-SBL.hex -O CC2530ZNP-with-SBL.bin`

On both platforms this fails with `Invalid bfd target`

Flashing the 2530 with the regular Zstack bin works fine using ccloader, and ECU works fine, reports Zigbee is up etc, but of course that does not support the DS3.

My question : 
- would it be an idea to include the BIN as well in the distributed file?
- are there any specific convert flags that I may missing?

Thank you,

patience4711, 2022-10-17:
You missed the -O binary flag. 
I renamed the ds3 file to from CC2530ZNP-with-SBL.hex to CC2530ZNP-with-SBL-ds3.hex
Now i tried it with **objcopy --gap-fill 0xFF --pad-to 0x040000 -I ihex CC2530ZNP-with-SBL-ds3.hex -O binary CC2530ZNP-with-SBL-ds3.bin** and it works.

in

silbor, 2022-10-17:
Confirmed from this side, thank you!

closed 2022-10-17 by silbor