patience4711/read-APSystems-YC600-QS1-DS3 #66: Does not connect to wifi

Erikkkkkkkkkkkk, 2022-11-05:
Hi,

I tried flashing my NodeMCU which went ok, but I ended up with:

Leaving...
Failed to leave Flash mode

Nevertheless, i get an AP ECU-xxxx and I do get the webpage at 192.168.4.1 where I can fill in my details. After filling in the wifi credentials the board flashes 3 times and is pingable on the new ip-adress in another range.

When visiting the IP at this moment I get the title: hansiart wificonfig but the screen remains blanc.

Ok, it says please wait until the led flashed, then click ok. I click ok.
It the show the IP address and tells me it is connected to my wifi, and it is still pingable.

I click close, it says it is gonna restart the device, but that's it. It is not longer pingable and the the default page: currently not connecting to a wifi network.

I tried another board, tried erasing everything with wifimanager, but i won't save my wifi credentials...

Any help is really appreciated :)

joffito, 2022-11-05:
Can you access it via a browser on an android device ? how did you try to flash the mcu ? if you have linux try flashing with the "esptool"

Erikkkkkkkkkkkk, 2022-11-06:
Thanks,

- I tried it several times through a Android phone and a Windows laptop,
both with different browsers.

- I flashed the mcu with the esp8266_flasher.exe included in your download.

- sadly, no linux here

patience4711, 2022-11-07:
The flash was successfull so that is not the problem here. You say you get the new IP address so at that moment your nodemcu is connected. When you click close then, after the reboot, the credentials should be remembered. Insteat of clocking close, you could try to powercycle it at that moment. 
Maybe you could try to connect the nodemcu to wifi with some other software and if that works, reflash it. Does your network address contain 192.168. etc?

Erikkkkkkkkkkkk, 2022-11-10:
Thanks, I tried all you suggested, but it did not help...

I flashed ESPhome on it with Home Assistant, which is working fine on wifi.
Then flashed again esp8266_flasher.exe but the same behaviour, does connect
to wifi, but just won't save credentials. Same with a Android hotspot.

My wifi's range is 192.168.178.x so that should not be interfering or
something.

Got any other ideas?

patience4711, 2022-11-12:
No sorry, i don't know what the cause is of this strange problem. I think the credentials actually are remembered but that after the reboot it takes too long to connect. After 10 attempts it is going to start the accesspoint.
I have another project that uses the same routine to connect to wifi. You could try to install that on your nodemcu and see if it works?
[led christmas star](https://1drv.ms/u/s!AkMG5FvUwhedigTYYQ75S09zNV_d?e=bd7Lsh)

frtz13, 2022-11-24:
did you try to completely erase the flash memory (using esptools, for ex.), before flashing the ESP again?

Erikkkkkkkkkkkk, 2022-11-25:
No, completely forgot about that, and it worked! thanks a lot

I used Arduine IDE to erase erverything and it worked straight away after that.

Now I just need to get the right firmware on my CC2530. I can backup the firmware, i can erase it, but it hangs when flashing, i just hangs on line 1/128. Tried with nohub as I am using Putty, same thing. I don't have the right cables or powered usb to get a keyboard and display working on the Pi Zero.  Maybe i can find them somewhere next week...

I followed this guide btw: https://blog.boris-wach.de/permalink/265

Any ideas for now?

frtz13, 2022-11-25:
I did it following this guide: https://zigbee.blakadder.com/flashing_ccloader.html, and it worked.

Erikkkkkkkkkkkk, 2022-12-03:
Thanks again, this worked. :)

I got everything working now, i paired 3 DS3 inverters, got 3 different status codes, but only 2 are connected at one time. I've read this is a known problem, is there any fix or should I just run 2 instances?

Erikkkkkkkkkkkk, 2022-12-06:
I got all 3 inverters working now... don't know exactly why. The day before yesterday i made sure i got a decent zigbee signal quality. Tried to pair again, did succeed but no data. I power-cycled the inverter, paired again, no data. But today i noticed unpaired, instead of the ID. So I paired again, and now they all work. :)

Thanks again for this project @patience4711 and all other folks who contributed. I'll neuter a dog if https://neuteringmarket.eu/ as requested in the readme.txt with the zigbee-firmware. If the site is working again that is.

closed 2022-12-06 by Erikkkkkkkkkkkk

Erikkkkkkkkkkkk, 2022-12-06:
Please rest asured that doggo Monty can fuck as much as he likes, well, in a few days i guess...

https://neuteringmarket.eu/images/R0_20221205115433_5d6b2d262233de.jpeg