patience4711/read-APSystems-YC600-QS1-DS3 #68: Support for Ap Systems YC500i

gerardwr, 2022-11-22:
Hi,

Job well done!

Is this software also suitable for the older YC500i?

Regards,
Gerard.

patience4711, 2022-11-23:
Hi Gerard. Sorry but no.

gerardwr, 2022-11-23:
That’s too bad. Does this mean that the YC500i has a different Zigbee protocol, or that it does not use Zigbee at all?

patience4711, 2022-11-23:
You tell me, you own one, i don't.

gerardwr, 2022-11-24:
Yep, own one since yesterday!. No clear info found on Google, So I hoped you could give me a head start. Let’s see what I can find out. Thanks for your reaction.

gerardwr, 2022-11-25:
Found some more info on the YC500. It seems that the data from the inverter is sent by pulses on the 220V power line. So no Zigbee communication on the YC500.

Some technical info here:

[https://sites.google.com/site/ecu2pvo/aps-aflc](url)

closed 2022-11-29 by patience4711