patience4711/read-APSystems-YC600-QS1-DS3 #71: Pairing to any inverter fails, YC600 and DS3-L

emmrichd, 2022-11-26:
Hello,

I own two yc600 and one ds3-l
I used the firmware from the forum link, i.e. the "CC2530ZNP-with-SBL.hex", is that the right one?
For the serial console of my esp, the check of the zigbee module seems fine.
However, when trying to pair, all I get is:
`<html>
<body>

26-8:51:1 | pair | failed, inverter got id 0x0000
-- | -- | --
26-8:50:45 | zigbee | zigbee module reset
26-8:50:44 | pairing | start pairing inverter sn 7030000450xx
26-8:50:43 | pair | failed, inverter got id 0x0000
26-8:50:27 | zigbee | zigbee module reset
26-8:50:26 | pairing | start pairing inverter sn 7030000450xx
26-8:50:4 | pair | failed, inverter got id 0x0000
26-8:49:48 | zigbee | zigbee module reset
26-8:49:47 | pairing | start pairing inverter sn 4080001406xx
26-8:49:46 | pair | failed, inverter got id 0x0000
26-8:49:29 | zigbee | zigbee module reset
26-8:49:28 | pairing | start pairing inverter sn 4080001415xx
26-8:49:27 | pair | failed, inverter got id 0x0000
26-8:49:11 | zigbee | zigbee module reset
26-8:49:10 | pairing | start pairing inverter sn 4080001415xx

</body>
</html>`

Hay onyone an idea, what is going wrong?

Regards, Emmrich

emmrichd, 2022-11-27:
Hello,

I got partial success now. I disconnected all inverted from the mains during the night for some hours.
So today I was able to pair all inverters:
`inverter 0 serialnr 703000045070 ID = 0xD8A3 signal quality: n/a
inverter 1 serialnr 408000141569 ID = 0x5713 signal quality: 36.1%
inverter 2 serialnr 408000140662 ID = 0x5713 signal quality: 36.1%`

However, I get not data from the DS3-L.

Any idea how to continue, please?

patience4711, 2022-11-29:
Hi, where did you get the firmware for the zigbee? What is the forum link? You should use the firmware included in the package you downloaded, in the folder "CC25xxfirmware ds3" , i hoped that that was obvious. You didn't mention what zigbee module you use.

emmrichd, 2022-12-01:
I got it from the discussion which is being linked to on the main page: [https://github.com/Koenkk/zigbee2mqtt/issues/4221](url)
I used this module [https://www.ebay.de/itm/284110560174](url) which seems to have a very low range. 2m away from my yc600 inverters, I get about 34% signal strength. So it just might be a range issue, as my ds3 is further away. I now bought a H-2 Z-002 (TICC2530 CC2591)  module and give it a try over the weekend.

emmrichd, 2022-12-04:
I just installed the new Zigbee module, about 1.5m away from the inverter, I now even get worse signal quality?!?

PAIRED INVERTERS
inverter 0 serialnr 703000045070 ID = 0xD8A3 signal quality: n/a
inverter 1 serialnr 408000141569 ID = 0x5713 signal quality: 0.8%
inverter 2 serialnr 408000140662 ID = 0x5713 signal quality: 0.8%

How can this happen, please?

patience4711, 2022-12-05:
No sure what the cause could be. You have 2 inverters with the same id? This should be impossible. The signalquality figure comes within the poll answer so it is only available if you can succesfully poll your inverters. Maybe wise to start with only one of the yc600's to see if you can make that work.

closed 2023-01-03 by patience4711