patience4711/read-APSystems-YC600-QS1-DS3 #72: Pairing QS1 and YC600 inverters fails

JK0ning, 2022-12-07:
Hi there,

I am experiencing some issues when trying to pair my 2 QS1 (each four panels attached) and 1 YC600 (2 panels attached) inverters to the ECU. Pairing an inverter results in the following:

'<html>
<body>

init zb coordinator
zigbee module reset
zb send cmd 0
inMessage = FE064180010200020603C3FE0166050062 rc = 17
zb send cmd 1
inMessage = FE064180020200020603C0 rc = 11
zb send cmd 2
inMessage = FE0166050062 rc = 6
zb send cmd 3
inMessage = FE0166050062 rc = 6
zb send cmd 4
inMessage = FE0166050062 rc = 6
zb send cmd 5
inMessage = FE0166050062 rc = 6
zb send cmd 6
inMessage = FE0164000065 rc = 6
zb send cmd 7
inMessage = FE00660066FE0145C0088C rc = 11
trying pair inv 0
pair command = 2524020FFFFFFFFFFFFFFFFF14FFFF140D0200000F1100801000082359FFFF10FFFF80971B01A3D899
sending paircmd 2
received : FE016402C2A5FE0145C0098D
pair command = 1A24020FFFFFFFFFFFFFFFFF14FFFF140C0201000F0600801000082359D7
sending paircmd 3
received : FE0164020067FE034480001401D2
not found serialnr, returning
pair command = 2524020FFFFFFFFFFFFFFFFF14FFFF140F0102000F1100801000082359A3D810FFFF80971B01A3D8E1
sending paircmd 4
received : FE0164020067FE034480001402D1
not found serialnr, returning
pair command = 1A24020FFFFFFFFFFFFFFFFF14FFFF14010103000F060080971B01A3D84F
sending paircmd 5
received : FE0164020067FE034480001403D0
failed, inverter got id 0x0000
sending NO cmd
inMessage = FE0164010064FE034480001401D2 rc = 14
inMessage = FE0E670000FFFF80971B01A3D8000007090011 rc = 19
found 0709, running oke

</body>
</html>'

Does anyone know what went wrong? I was pairing while the sun was out.

Thanks in advanced!

altbau9944, 2023-01-22:
Hi,

I am facing the same problem.
Have you been able to air your devices already??

Cheers
altbau

JK0ning, 2023-01-22:
Hi,

Sadly I have still not aired my devices yet... Must say, I haven't tried it since posting this.

Greetings,

Jur

masp31, 2023-01-29:
Hi.

I have the same problem, when try to pairing. 
....
failed, inverter got id 0x0000
,,,,
I have already confirm the SN. And I  was pairing while the green led in the YC600 was on.
I have test all the connections, the MQTT and the ZigBee 
I dont underestand...

Best 

Miguel Pereira

patience4711, 2023-01-31:
The program tries to find the serial nr in the pairing answer, when found the pairing succeeded. In your cases if fails. Impossible to say why the pairing is sometimes difficult. After multiple attempts it will work provided that there are no errors in the wiring/settings and the coordinator can be started normally. Sometimes after a reboot of the inverter or a reboot of the esp-ecu it suddenly works.

closed 2023-01-31 by patience4711