patience4711/read-APSystems-YC600-QS1-DS3 #73: CC2530 no serial communication

Tombost, 2022-12-13:
Hello,

I have a nodemcu V3 (V0.1) I have flashed the ESP-ECU-v9_6 firmware with the esp8266_flasher.exe software.
I have a CC2530, brand new from aliexpress. I have flashed the CC module with a raspberry. I have flashed DS3 firmware : CC2530ZNP-with-SBL.hex
The result was good. If I use the cc_dhipid I got ID=a524. So my zigbee module is working.
And flash process take 3 min (reading line 15490 / file loaded 15497 lines read / wrinting page 128/128 / Flash OK)

But when I connect it to the esp the serial communication is not good. I use simple jumper wire (same for the RPi).
I connect like this : 
CC2530 TO ESP
GND TO G
VCC TO 3.3
RST TO D5
P03 TO D8
P02 TO D7

I have try to reverse D7 and D8 but same result : 
If I send : 10;zbt=2710AABBCC
I got :
going to send a teststring, len=17
sendCmd = FE032710AABBCCE9
answer

What I am doing wrong ?

[image]

patience4711, 2022-12-16:
Hi, impossible to say whats wrong. Your hardware seems oke. Did you use the cc2530 firmware from the downloaded package? 
The zigbee should be initialized first, this happens at boot but you can also do this via the console. Powercycle with the wires connected should initialize the module. So the only thing to suspect is the wiring, maybe one of the wires got faulty after flashing.  Otherwise i wouldn't know, sorry.

closed 2023-01-12 by patience4711

altbau9944, 2023-01-19:
> Hello,
>
> I have a nodemcu V3 (V0.1) I have flashed the ESP-ECU-v9_6 firmware with the esp8266_flasher.exe software. I have a CC2530, brand new from aliexpress. I have flashed the CC module with a [...]

Hi,

Have you been able to solve your problem?
I am facing the similar issues, with the same components

Cheers
altbau

ghmpi, 2023-02-08:
I seem to have the same problem.. but with the CC2531 module. Did you folks ever fix your problems?

going to send a teststring, len=17
sendCmd = FE032710AABBCCE9
answer

Whatever I do I can't seem to get the ESP to talk to the zigbee module. I am using the standard CC2531 board that is selling everywhere and flashed it successfully with flash_cc2531 on a Pi. I flashed it with the file CC2531ZNP-with-SBL.hex

I looped back the serial ports on the (D7/D8) on the ESP and it is getting it's commands echo'ed back to it, so the serial port on the ESP appears to be working.

I also confirmed that The top two pins on the header row of the CC2531 are going to P02 and P03 on the actual CC2531 chip itself. cc_chipid on the Pi also reports an ID = b524., which the documentation says is correct.

I tried both ESP-ECU_v9_5 and ESP-ECU_v9_6

errorCode is still reporting as 10 and zigbeeUp=11

I assume something is wrong with the zigbee module or the code on it. It blinks it's green LED occasionally and the led turns on and off when doing a health check on the ESP. I tried different power supplies and different hookup wires. I can't seem to figure it out. I think I have to connect the module to some other device and see if I can talk to it. Anyone know a quick command I can send the zigbee device firmware so it talks back to me? Anyone know the baud rate?

Thank you for your time.

Regards,
-Moses