patience4711/read-APSystems-YC600-QS1-DS3 #74: Further development

joffito, 2022-12-25:
Hi there,

i was wondering if this project will receive some further development ?
Currently i am struggling a bit with logging the data from the MQTT and displaying it.
What tools / hardware do you guys use to log and display the data? Are there maybe any nice tutorials out there ?

For me it would be sufficient to have the total amount of energy per day accumulated, stored (in the EEPROM) and shown in the main page of the ESP. Maybe also the sum of both panels would be "nice to have" (ok, the sum here is usually very easy to calculate ;-))

I noticed a typo when querying the ESP like so: http://192.168.178.83/get.Inverter?inv=0

It says avc which should probably be vac (voltage ac).

joffito, 2023-01-07:
Again, as someone new to the IoT world i installed Domoticz on my RPI. Can i somehow log the data from the ESP ECU with Domoticz ? If yes, which type of hardware is used during setup of hardware in Domoticz ?

patience4711, 2023-01-09:
This is a question for the domoticz knowlwdge base. But yes, the hardware to be installed is mqtt-client-gateway. remote address: localhost (if you have mosquitto running on your RPI) port 1883.
Setup a sensor of the type general, kWh.

joffito, 2023-01-09:
Yeah, i think i need to process the message with NodeRed and then pass it to Domoticz via a MQTT Device and Dummy Sensor. Once i figured out how it works i'll post it here in case someone wants to go the same route.

What do you think about summing up the energy generated per day (when the ESP goes to night mode) and displaying it as an increasing value on the page with the panel data ?
And displaying the sum of both panels as well ?

I tried to compile and do it myself but for some reason it complains at some Time.h dependency in the sunmoon lib. Once i got the first thing to work i'll move on to the next ;)

patience4711, 2023-01-12:
Hi, 
No need to process the message, in the options as to the mqtt format, the 1st format works with domoticz. You can setup a sensor for every inverter. If you want a sensor per panel than you need to process one of the other formats. 
The typo should be acv, in future releases (if any) it will be corrected. Compiling with board version Esp8266core 2.7.1 should work.  As to compiling yourself you are on your own, i don't support that, costs me too much time.

Keeping the total energy stored would mean that this value must be written to eeprom after every poll. And that for every inverter. This might wear out the eeprom so i don't like this very much. The size of the program is such that there is almost no space left to make ota updates possible. So the possibilities to extend the program are limited.

If you want more options you better go to the raspberry zero implementation. 

Good luck and happy programming.

joffito, 2023-01-13:
OK, so the IDX is of course defined in the Inverter Setup and as soon as i changed it to the IDX of Domoticz it works now and with Format 1 is shows the total of the current power production of my panels.
Great !

If somebody knows how to sum up the total power produced for Domoticz any input is appreciated (NodeRED etc.).

Anyhow thanks for the patience (pun intended ;))

patience4711, 2023-01-13:
Indeed, you want to be sure that the value is correct and crash proof. In the raspberry implementation, the values are stored in a database which gives the possibility to analyze a defined period for all kinds of statistic info. Complete with the charts. In case something goes wrong, the values can be corrected. I would recommend buying a raspberry zero.

joffito, 2023-01-13:
I think a common practice for kWh meters is to the the value and store it immediately in the EEPROM. then check if the new value is bigger than the old from the eeprom and if yes move the position up for 4 bytes of a float and then write again etc. with 4 kb of EEPROM on the 8266 you have 1024 "positions" with each 100.000 write cycles. plenty to write a value every 5 min.
also i do not want to buy an RPI at the moment as they are very hard to get and i don't want to consume more and buy a device for my hopefully sustainable solar power stuff. i had the ESP and the CC2530 laying around anyways..

anyhow, today some of my brain cells gathered and let me know the solution with domoticz:

- Create a hardware of type MQTT Client Gateway with LAN interface and enter your MQTT broker info
- Create a hardware of type Dummy and use this to create a virtual sensor
- For the virtual sensor use type Electric (instant + counter)
- Check the IDX of this device and enter it in the ECU ESP in the inverter setup as the IDX
- It shows current power consumption and logs the total kWh --> this is what i was looking for

And yeah, then the ESP ECU crashed the first time on me today. The values of the panels simply would not update. A reboot of the ESP didn't help and neither did a power cycle.
Then i AC power cycled the inverter and the ESP showed plausible values again.
Maybe there is a bug in the firmware of the inverter or it send some packets which made the ECU get a hickup.

Would be interesting to know if the Raspberry Pi and ESP implementation use the very same communication algorithm with the inverter and if the values from the panels get stuck on the RPI implementation as well from time to time.

patience4711, 2023-01-20:
The communication to the inverter uses a protocol, rpi or esp not important. It happened to me on a dark day that the inverter didn't provide any values, the chart of the power was a sharktooth. it seems that the inverter keeps resetting and starting at null. Next day all was normal again.  No idea what this was. Anyway, with the RPI you can put a value in the database that represents the expected energy for that day so in the end not much harm done.

joffito, 2023-02-09:
I just wanted to add that it would be cool if you could compile the 9.6 v again with the healtcheck done automatically upon start. it was commented out and then compiled and thus the zigbee does not start any more automatically.

Closing this topic now anyhow

closed 2023-02-09 by joffito