patience4711/read-APSystems-YC600-QS1-DS3 #78: How to restet the admin password

LordDarkneo, 2023-01-20:
Hello There,

I was following the wiki to change the "user" account password in the Settings tab... But unfortunately it seems to have changed the admin password also...

After saving, I have been requested the login page below
[image]
And I cannot connect using admin pwd 123456789... I tried to keep admin with the password I set up, still not working... Even user (instead of admin) is not working with both passwords...

I tried to reflash the nodemcu, but it did reset the configuration (I still have the same message).

How to reset the admin password? 

Thanks for your help!

patience4711, 2023-01-20:
The admin password is set in the accesspoint for security reasons. Via the menu you can start the AP.

LordDarkneo, 2023-01-21:
> The admin password is set in the accesspoint for security reasons. Via the menu you can start the AP.
The problem I am facing is that I cannot access the access point no more. 
I tried to connect D4 and ground, but it doesnt drop the wifi configuration (the board keeps booting normally). I tried to flash the board using a blank firmware and reflashing with the ESP-ECU, but the config is still.... And I cannot access the menu no more because of the admin login console...

joffito, 2023-01-21:
power cycle the esp and then hold the flash button for a while. it'll reset it (or at least the wifi settings).

LordDarkneo, 2023-01-21:
Thanks, that did the trick! Press the Flash button, the led lights up. Keep it pressed until the led is off and release the flash button.

The "root" issue was my password too long.... it was truncated in the interface (you see the limitation to 10 digits only when the field is empty... Could be good to write it in the documentation as a warning, no?)
Thanks a lot for your support. Waiting a bit of sun to try this on my DS3L :)

closed 2023-01-21 by LordDarkneo