patience4711/read-APSystems-YC600-QS1-DS3 #79: compile error in arduino project

smartboart, 2023-01-23:
All files with big letter  
#include "AAA_MENUPAGE.H"
#include "PORTAL_HTML.H"
#include "HTML.H"
will throw this error.
ESP-ECU_v9_6:30:18: fatal error: HTML.H: No such file or directory
 #include "HTML.H"
compilation terminated.
exit status 1
HTML.H: No such file or directory

When theses files will be renamed to xxx.h and change the lines ESP-ECU_v9_6.ino file , then this errors are gone and the next error popps up.
'Mqtt_Enabled' was not declared in this scope
When i now change the line under // variables mqtt ********************************************
 in the ESP-ECU_v9_6.ino file to ( bool    Mqtt_Enabled =  true; )
then all the compile errors are gone and i can export the compiled bin file.
But this bin file is then not running correctly.

Can you tell me how you have complied the actual bin file with this master branch?

Tank you very much.

smartboart, 2023-01-23:
with this change after reboot the mosquitto setting format is always disabled. So the change is still not good. 
There is something wrong with the bool 'Mqtt_Enabled' ...
It was disabled //bool    Mqtt_Enabled =  false; ..
How can i compile the sketch with diabled //bool    Mqtt_Enabled =  false;
Thanks for feedback..

patience4711, 2023-01-23:
Sorry, it is really impossible to know why it won't compile for you. Anyway, the bool Mqtt_Enabled is not used, this should stay commented out.  You can enable mosquitto by choosing an option in the mqtt format box. The help button explaines this.

smartboart, 2023-01-24:
Thanks for fedback...
Can you tell me which Arduino Version you use?

smartboart, 2023-01-24:
Ok on my collegues PC it was working. He same version like me and we did the same togehther.. strange.. But issue will be closed anyway...
Thanks

closed 2023-01-24 by smartboart

swbouman, 2023-02-02:
Goodmorrning @smartboart ,

I tried to compose the arduino file for this project but i'm running in a lot of errors. A bunch of return errors, and 'not declared in this scope'

Can you pls point me in the right direction?

smartboart, 2023-02-02:
You need to Import the missing include the missing xxx.h files...if not available Google and Import direct the zip files from github..Rename big xxx.H files  to xxx.h and also change in the ino file..