patience4711/read-APSystems-YC600-QS1-DS3 #80: DS3L compatibility confirmed

LordDarkneo, 2023-01-25:
Hello there,

Just for you to know: I have been successfully pairing the CC2530ZNP-with-SBL.hex (from the DS3 folder) with my brand new DS3L (connected to 2 solar panels).

My DS3L is 703000354000 serial number (not the real SN, but just to let you know that it's the new version after the issue they faced few months ago). Weather is very cloudy for the next days, so I hope I'll be able to confirm the behavior soon (I also integrated it in domoticz).

I also made a quick video (in french) to help people configure CC2530 and Nodemcu, I can share it if you're interested (with my own3DPrinted case ;) )
Regards

swbouman, 2023-01-25:
Dear @LordDarkneo,

Thank you for your review. I am eager to view the video. My panels will be installed at the end of February. I have already installed the software on the ESP and the system is currently running. Unfortunately, I am unable to test it at this time.

I hope that my inverters will be able to connect with each other. The farthest inverter will be approximately 25 meters away from the CC2530. Can you confirm that the system will be able to detect the inverter at that distance? If not, I will need to find a way to place a Zigbee router closer.

Best regards.

LordDarkneo, 2023-01-26:
Hi and thanks for your interest :)

The video is still in progress. To answer your question: my actual DS3L is like 10m far from my ESP-ECU (separated by a windows, not a wall) and my signal is around 10%
[image]

I guess 25m will be a bit far....

swbouman, 2023-01-26:
Thank you for your prompt reply. 

I have recalculated and found that the panel that is farthest away is approximately 13 meters away. I plan to mount the ESP-ECU outside, so I need to find a waterproof enclosure to install everything in.

I hope this will be sufficient.

It's also exciting to think about whether it will be possible to communicate with 7 inverters. Only time will tell.

By the way, do you use more than one inverter?

LordDarkneo, 2023-03-25:
Hi and sorry for late answer. 
As I am using Domoticz, I only use the format 1, it is barely enough to me. And yes I only have one inverter, with 2 solar panels. But unfortunately, even if the communication is ok, it does not produce... I sent it back to the seller for upgrading it...

And just in case someone wants to give a try with my box: https://drive.google.com/drive/folders/15VWAfAMM3C682tL0pfZBzZpJcrlixr_v

patience4711, 2023-04-06:
@LordDarkneo Nice box. Can you tell me what Esp board fits in your box? Then i can upload your file to the repo to make it available to everybody.

closed 2023-04-14 by patience4711

LordDarkneo, 2023-04-14:
> @LordDarkneo Nice box. Can you tell me what Esp board fits in your box? Then i can upload your file to the repo to make it available to everybody.

Hi, Sorry for late answer. 

I did use a CC2530 and a Nodemcu v3. there is very much space for the CC2530, I used a hot glue gun to keep the board in place, just at the opposite of the antenna.