patience4711/read-APSystems-YC600-QS1-DS3 #81: Pairing without Serial / Read serial number somehow?

derhappy, 2023-01-27:
I just finished flashing / building the reader and wanted to pair my (only) yc600 to the esp-ecu. Finally reading the related documentation (what I probably should have done beforehand) I realized I need to enter the serial number.
Unfortunately our inverter is mounted behind the panel on a wall... 
Is there any chance to read the serial without physical access to the inverter?

patience4711, 2023-01-27:
Yes, you need to build a zigbee sniffer [look here](https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html). The inverter sends a beacon signal that contains the serial nr. So you have to buy a cc2531.

derhappy, 2023-01-30:
Thanks for your quick response!
I was hoping to maybe do some console magic to somehow get the serial. I've been digging through some of the other implementations and the serial should be sent by the inverter every now and then as you pointed out. But I guess this is too low level to get it to show in the console in the current implementation.
Maybe that's an idea for some future version if you get very very bored at some point ;-)

At this point I'll just put this project in a drawer until I maybe some day get out the ladder to get to the inverter. It already took me way more time to set up than expected.

closed 2023-01-30 by derhappy