patience4711/read-APSystems-YC600-QS1-DS3 #82: Recompile ESP-ECU9.6.bin

swbouman, 2023-02-01:
Hey @patience4711,

ould you please recompile ESP-ECUv9.6 please? I tried to compile it myself, but i'm missing a lot of lib eg. #include "ESPAsyncWebServer.h" and SunMoon.h. The program will not compile in Arduino IDE, due to the fact sunMoon.h is not available in the arduino libs and "ESPAsyncWebServer.h" is a custom lib, which is not available in your rep. Installing a lib which looks same spelled as yours throws gigantic errors on comping.

thanks in advance!

patience4711, 2023-02-01:
The compiled binary ESP-ECU-v9.6 can be found in the list of files. I can't see why this should be recompiled.

swbouman, 2023-02-01:
Because there were changes in the code, you fixed some typos etc. The nodemcu bin had been updated with the change off the files. The esp-ecu bin wasn’t updated since 4 months. 

hope you can point me in the right direction fox fixing above mentioned libs. 

I appreciate your project!

swbouman, 2023-02-01:
[image]

h4nk0r, 2023-02-01:
> [image]
> 
> 

I think its the last one in your screenshot

swbouman, 2023-02-02:
@h4nk0r i’ll install that one. Thnx!

patience4711, 2023-02-02:
@swbouman  
To be sure you have the most recent version i recompiled the program today, you can [download it here](https://github.com/patience4711/read-APSystems-YC600-QS1-DS3/blob/main/ESP-ECU_v9_6.bin)

swbouman, 2023-02-03:
@patience4711 Thank you! I updated the Wemos successfully!

joffito, 2023-02-03:
I flashed the "new" 9.6 version but my ESP could not get past the "initializing zigbee" procedure. it would display this red message forever. Could not get it to work.
Then i flashed the "old" 9.6 and it works again. 
Anybody else ? Maybe because of the change in zigbee coordinator.ino ?

swbouman, 2023-02-03:
@joffito This is correct. I thought it was because I flashed NodeMCU to a Wemos. But no difference. The coordinator will not get a start command during boot. So the red text will stay until you tell the ECU to start a health check. Afterwards the coordinator is started and functioning. 

@patience4711 commented out the health() command in the boot sequence and added a 10 sec pause. Probably a typo?

swbouman, 2023-02-03:
ZIGBEE_COORDINATOR.ino:

`    coordinator_init();
     if(normal) sendNO();
     // now check if running
     //healthCheck();
     delay(1000); // to give the ZB the timne to start
     if ( checkCoordinator() == 0 ) // can be 0 1 or 2
     //if ( zigbeeUp == 1 ) 
     {
`

patience4711, 2023-02-15:
Conclusion: These difficulties were caused by a test flag that I forgot to remove at compile time. 
It was a rush job and then mistakes can happen.. sorry

closed 2023-02-15 by patience4711