patience4711/read-APSystems-YC600-QS1-DS3 #83: credentials prompt on latest version

kersvanbinsbergen, 2023-02-03:
Hi,

I'm just getting started. I flashed a d1 mini pro. It's connected to my network so it has an ip address within my subnet. But when i connect it askes for credentials.

I tried 123456789, 0000 in the password field. With blank username and also with username admin.

Am i missing something?

joffito, 2023-02-03:
you set up the credentials during wifi setup. i think the default user is admin and then your custom pw

kersvanbinsbergen, 2023-02-03:
Really?

I didn't see that. I'm gonna check. Thanks

Groeten, Kers

Verzonden met Proton Mail mobiel

-------- Oorspronkelijk bericht --------

kersvanbinsbergen, 2023-02-03:
Sorry guys. It was admin:0000

kersvanbinsbergen, 2023-02-03:
Are the pin numbers in the wiki shown in "wiring diagram ESP-ecu cc2530" the same as for the d1 mini pro?

Wires are correct but i get no answer when i type 10;zbt=2710AABBCC

patience4711, 2023-02-03:
I wouldn't know, think so but you can google that yourself. If it's not working please check out the numerous issues already existing.