patience4711/read-APSystems-YC600-QS1-DS3 #84: Pairing failed

ThibautBascoulergue, 2023-02-08:
Good morning,

I can't pair my micro inverter. Can we know what is the difference between the three Firmware CC2530? Which one to take. THANKS

with regard to the various other issues, I tried to disconnect the panels so that the micro inverter restarts and tried a new pairing but it does not work.

Here are the logs:

<html>
<body>

inMessage = FE0166050062 rc = 6
--
zb send cmd 3
inMessage = FE0166050062 rc = 6
zb send cmd 4
inMessage = FE0166050062 rc = 6
zb send cmd 5
inMessage = FE0166050062 rc = 6
zb send cmd 6
inMessage = FE0164000065 rc = 6
zb send cmd 7
inMessage = FE00660066FE0145C0088CFE0145C0088CFE0145C0088C rc = 23
trying pair inv 0
pair command = 2524020FFFFFFFFFFFFFFFFF14FFFF140D0200000F1100705000005150FFFF10FFFF80971B01A3D85A
sending paircmd 2
received : FE0164020067FE0145C0088CFE0145C0088CFE0145C0088CFE0145C0088CFE0145C0098D
pair command = 1A24020FFFFFFFFFFFFFFFFF14FFFF140C0201000F060070500000515014
sending paircmd 3
received : FE0164020067
not found serialnr, returning
pair command = 2524020FFFFFFFFFFFFFFFFF14FFFF140F0102000F1100705000005150A3D810FFFF80971B01A3D822
sending paircmd 4
received : FE0164020067
not found serialnr, returning
pair command = 1A24020FFFFFFFFFFFFFFFFF14FFFF14010103000F060080971B01A3D84F
sending paircmd 5
received : FE0164020067
failed, inverter got id 0x0000
sending NO cmd
inMessage = FE0164010064FE034480001401D2 rc = 14
inMessage = FE0E670000FFFF80971B01A3D8000007090011 rc = 19
found 0709, running oke

</body>
</html>

joffito, 2023-02-09:
Which inverter are you using ?

In case it's a 3DS-S i managed to successfully pair with firmware CC2530ZNP-with-SBL.hex

ThibautBascoulergue, 2023-02-11:
Hello, I have resolved my problem with the correct c22530 firmware with 2591 :)

closed 2023-02-11 by ThibautBascoulergue