patience4711/read-APSystems-YC600-QS1-DS3 #85: EcuESP reports negative energy values

joffito, 2023-02-09:
I observe that during midnight my Domoticz on my RPI seems to receive a negative energy value from the ECU ESP.
Unfortunately i do not have more data to show because i set it up from scratch yesterday but i observed this phenomenon since more days.

I set up my domoticz device as Electric (Instant + Counter) with the values from the picture.
I think somehow during reset of the daily Energy generated the ECU ESP seems to send a negative value over MQTT.

Can someone confirm this ?

[image]

[image]

bwbboot, 2023-02-10:
I have this same issue. At first i thought this was ESP ECU related but I see the same behaviour using the latest RPI ECU.

bwbboot, 2023-02-13:
It seems that using the option energy "read from device" in Domoticz is a total counter and calculates the daily delta. However the values from ECU are reset daily causing the negative value. I did some testing and changing to "Computed" solves the issue. 

@patience4711  is there an option to keep increasing the counter without a daily reset?

joffito, 2023-02-15:
you're right, thanks!
this is domoticz behaviour. i changed to computed and the issue is gone.
closing the issue now.

closed 2023-02-15 by joffito