patience4711/read-APSystems-YC600-QS1-DS3 #87: DS3 not pooling

n000n000, 2023-02-10:
Hi,
I just finish my setup using a Nodecmu v3 (flashed with ESP-ECU_v9_6.bin) connected to a CC2530 (flashed with CC2530ZNP-with-SBL.hex)
I have 3 DS3 inverters with 2 panels connected on each.
Everything wents fine, even the pairing with the inverters.
Unfortunately, I don't have any polling.

This is the result when I send 10;HEALTH
_check zb system
inMessage = FE02610179071C rc = 7
ping ok
inMessage = FE0E670000FFFF80971B01A3D8000007090011 rc = 19
found 0709, running oke_

Any idea what can cause the issue ?
Thanks

n000n000, 2023-02-10:
I manage to have each inverter working separetely but as INV 0.

But as soon, as I use a 2nd or 3rd one, it doesnt work.
When I do a **10;POLL**  on the 2nd inverter, I always get

_no AF_DATA_CONFIRM
polling failed 11_

Is it a known limitation for the DS3 ?
Should I do 3 differents hardware setups for each inverter ?

n000n000, 2023-02-10:
Update:
With version 9.5, I manage to have 2 inverters working but not 3 at the same time.
Weird...

swbouman, 2023-02-10:
How did you create the 3rd one twice? 

I’m not able to compile the code. So I can’t make any additions or workarounds. 

I’ll get 7 inverters. I hope I can read data from all 7 inverters with this program. 

I really like the project but it’s a pity that it’s sort of closed source, due to the fact the program is not compiling due to dictionary depencies.

n000n000, 2023-02-10:
I just add the 3rd inverter -> not working
Then I create a 4th with the exact same parameters -> not working but the 3rd start to work :-)

patience4711, 2023-02-10:
You must be doing something wrong, please check how many inverterfiles you have. Maybe wise to start all over with a clean sheet and add/pair the inverters one by one.

ThibautBascoulergue, 2023-02-11:
Good morning,

 I have the same problem. I have a DS3 micro inverter which is well paired but no automatic polling. If I don't force manual polling, the values ​​don't update.

patience4711, 2023-02-11:
@ThibautBascoulergue did you check the autopolling? Please see the wiki chapter polling.

ThibautBascoulergue, 2023-02-11:
Yes, autopolling is sticked

smartboart, 2023-02-12:
same issue here, pairing was done sucessfully but no values.
[image]

n000n000, 2023-02-12:
Did you try to use version 9.5?
For me it has resolve the issue and now I have the values for my 3
inverters.

patience4711, 2023-02-13:
I am sorry, in my latest compilation i forgot to remove a testflag, if you install https://github.com/patience4711/read-APSystems-YC600-QS1-DS3/blob/main/ESP-ECU_v9_6a.bin it should work oke.

Domidx is an identifier for "domoticz" and ds3 conversion an callibration value for ds3 inverters. (obsolete now)

n000n000, 2023-02-14:
Hi

I've updated from 9.5 to 9.6a and everything seems to be ok.
I also supress my "double inverter" I had to create to have 3 inverters working at the same time.
Thanks a lot for the update.

Now I have to understand how to use the MQTT information in HomeAssistant. 
Anybody info will be valuable as I discover this new protocol with this device.
Thanks

frtz13, 2023-02-14:
you may want to have a look at the (closed) issues #43 and #57

smartboart, 2023-02-16:
working fine now...thanks

closed 2023-02-17 by patience4711