patience4711/read-APSystems-YC600-QS1-DS3 #88: Can't talk to CC2531, no serial communication

ghmpi, 2023-02-11:
Hello,

I'm trying to get this running on, but so far have not have luck with the esp talking to the CC2531 module. 

Someone else had a similar problem with a CC2530 module, and I posted there, but the topic is now closed so I don't think anyone sees it anymore. That topic is here and my message is attached below to this post
https://github.com/patience4711/read-APSystems-YC600-QS1-DS3/issues/73

I did recently get another CC2531 board from a different vendor.. flashed it and tried that as well, same problem.
Am I reading the source correctly in that the CC2531 communicates via serial at 115200? Is there a simple command I can send it to test communications? I would like to hook it up to a Pi or some other device and manually check the serial communications on something other then the esp.. maybe try to isolate the problem to either the ESP of the CC2531 device. I don't know what else to try at this point.

Anyone have any advice on what else to check? Thank you for your time!

Regards,
-Moses

I seem to have the same problem.. but with the CC2531 module. Did you folks ever fix your problems?

going to send a teststring, len=17
sendCmd = FE032710AABBCCE9
answer

Whatever I do I can't seem to get the ESP to talk to the zigbee module. I am using the standard CC2531 board that is selling everywhere and flashed it successfully with flash_cc2531 on a Pi. I flashed it with the file CC2531ZNP-with-SBL.hex

I looped back the serial ports on the (D7/D8) on the ESP and it is getting it's commands echo'ed back to it, so the serial port on the ESP appears to be working.

I also confirmed that The top two pins on the header row of the CC2531 are going to P02 and P03 on the actual CC2531 chip itself. cc_chipid on the Pi also reports an ID = b524., which the documentation says is correct.

I tried both ESP-ECU_v9_5 and ESP-ECU_v9_6

errorCode is still reporting as 10 and zigbeeUp=11

I assume something is wrong with the zigbee module or the code on it. It blinks it's green LED occasionally and the led turns on and off when doing a health check on the ESP. I tried different power supplies and different hookup wires. I can't seem to figure it out. I think I have to connect the module to some other device and see if I can talk to it. Anyone know a quick command I can send the zigbee device firmware so it talks back to me? Anyone know the baud rate?

Thank you for your time.

Regards,
-Moses

patience4711, 2023-02-13:
The wiki points out that you can use the firmware for the cc2530. https://github.com/patience4711/read-APSystems-YC600-QS1-DS3/issues/45

If that doesn't solve the problem it's impossible to say whats wrong. Via the console you can send commands to the ZB module. The quick command is the loopback command FE032710AABBCCE9.  If you have a pi you could try the RPI-ECU image and test it with that.

ghmpi, 2023-02-13:
Patience4711,

Thank you for your reply! After your message I tried the CC2530 firmware and it worked!
I flashed my CC2531 with the CC2530ZNP-with-SBL.hex file in the \ESP-ECU-9_6\CC25--firmware-yc-qs\ FOLDER

Why would one provide the CC2531 firmware when the CC2530 one works? I'm confused.

One more small issue, I think it's related to the timezone offset. I can't seem to get auto polling to work. On my main page I always have "No inverter polling until sunrise!". The only way I got data from my inverter was to set auto polling to off and use the http interface to force a poll (http://ip_of_esp-ecu?POLL=0)

It seems to me that the timezone variable may be having trouble with negative values. I'm in PST, which is 8 hours behind UTC, so I put -480 in the timezone field. I also tried 0 and 480 (positive) values in the timezone field. I've attached some output from the info page below.. I believe "now" has to be between "switchonTime" and "switchoffTime", which is not happening!

Thank you again Patience4711 and any help you can provide. Your project here is pretty good.

Regards,
-Moses

=====================
timezone=0

value=0 inverterCount=1 zigbeeUp=1
pollRes= 300 now=1676320238 switchonTime=1676750655 switchoffTime=1676793855
unixtime=1676320238

=====================
timezone=-480

value=0 inverterCount=1 zigbeeUp=1
pollRes= 300 now=1676291480 switchonTime=1676750655 switchoffTime=1676793855
unixtime=1676291480

=====================
timezone=480

value=0 inverterCount=1 zigbeeUp=1
pollRes= 300 now=1676349097 switchonTime=1676837055 switchoffTime=1676880255
unixtime=1676349097

=====================

ghmpi, 2023-02-13:
I upgraded to 9_6a (btw, firmware version still says 9_6), still can't get the code to auto poll. It polls ONE time on first boot. Manual polling works.

It's 14:15, and it's daylight, but the code thinks it's night.. in the "ECU LOG" it states..
13-14:15:56 | system | start nightmode

Something seems to be wrong with the daylight, timezone or time code.

I have in my settings.. longitude = 34.000, latitude = -118.000, timezone = -480

Thank you for your time.

Regards,
-Moses

patience4711, 2023-02-14:
Hi Moses,
The timezone variable works with a negative value, that is not the problem. Is the system time correct? In the info page you can see if the time is retrieved from the ntp server. 
If i remember well there is a user in Sao Paulo whithout issues as to timing.  I think that the negative latitude value causes this, Your location is somewhere in Argentina?

ghmpi, 2023-02-14:
I removed the negative value on the latitude as a test and rebooted, still not polling. I now have 34.000 long and 118.000 lat.
I am in Los Angeles, and the system time is correct.

One thing that is suspect is the switchonTime... I have the following..

pollRes= 300 now=1676307884 switchonTime=1676664255 switchoffTime=1676707455
unixtime=1676307884

The unixtime and now is correct, it translates to Monday, February 13, 2023 5:04:44 PM, which is my current local time
BUT... switchonTime is Friday, February 17, 2023 8:04:15 PM, 4 DAYS LATER! This seems to me where the problem is.

Thank you!

Regards,
-Moses

ghmpi, 2023-02-14:
I think I figured out what was going on.. I had the LAT and LONG mixed up! Also, the longitude DOES take negative values, IE for Los Angeles 118 W is written as -118. If I omitted the negative sign the switchonTime would be wrong.

What confused me is that I have always written the latitude then the longitude, I believe this is convention. On the web interface, this is reversed. Yes, I should read it more carefully as well!

Will wait for sunrise and see how it works!

Thank you for your time patience4711!

Regards,
-Moses

patience4711, 2023-02-14:
@ghmpi 
I am sure it will work now.  I never wondered about the convention but in future release i'l have the fields reversed.

ghmpi, 2023-02-15:
@patience4711 
It's working great now!

Yes, convention is latitude first, I've written it this way for so many decades that it tripped me up!
Also, if you modify the documentation, a more obvious note that the CC2531 firmware may not work may be good.

Maybe soon someone will come up with an open source version of the firmware on the zigbee device. Instead of the hidden source version we have now, lol. In time.

Thank you for contribution!

Regards,
-Moses

patience4711, 2023-02-16:
Hi moses, 
In the wiki I now mentioned that both zigbee modules should be flashed with CC2530ZNP-with-SBL.hex. Can't remember why we have the CC2531ZNP-with-SBL.hex, there must be a reason for this. 
Btw there was an issue opened by someone who also reversed the coordinates, you missed that too. 
Anyway, glad that it works now, I hope you enjoy it a lot.

closed 2023-02-17 by patience4711